flandreas/antares #47: Renaming circuit when clicking breadcrumb

JanisFle, 2021-02-18:
Small suggestion concerning the renaming of a circuit: It would be convenient to have the ability to double-click the breadcrumb in order to rename the current circuit, which could be confirmed by pressing the enter key.
[image]

flandreas, 2021-02-19:
Thank you for this suggestion. The reason why you were missing this feature is probably bug #49 (you didn't see the property panel were you can change the circuit's name), and therefore tried to change it in the breadcrumb. Nevertheless, your suggestion is interesting, and I will look into it.

richardabendroth, 2021-05-19:
I would like to add that it would be more self-explanatory, if a circuit could be renamed in the explorer tree (like in the file system) rather than opening the circuit and change the Name property in the property editor, hit enter and save the circuit. Maybe additionally to the 'click a 2nd time after X ms to switch to rename mode' there could be an extra option in the right-click context menu of the tree node in the explorer tree.

closed 2025-04-23 by flandreas