flandreas/antares #57: Bi-directional Splitters/Combiners

richardabendroth, 2021-03-15:
Dear @flandreas, 

Antares is a great piece of work and despite the "Unexpected Error" glitches very fun to work with. However, one thing that is quite hard to work around is that the Splitter and Combiner components are unidirectional. This makes it particularly hard to model 8-bit tri-state bus transceivers like the LS74LS245A and use two of them on a 16-bit bus. Is there a plan to support bi-directional Splitters/Combiners any time soon? 

Best regards, 
Richard.

flandreas, 2021-03-15:
Dear Richard, thank you for your feedback. I understand your requirement concerning bi-directional splitters. The challenge is that tri-state-behaviour must be negotiated with all other output ports connected to the same net. Splitters in their actual implementation interrupt net connections. i.e. corresponding ports of the splitter belong to different nets, so no negotiation takes place.

I know that other comparable tools also struggle with this requirement. Nevertheless, I will think about it and try to implement it in release 0.4 (which is almost completed) or release 0.5.

Regarding "Unexpected errors": Antares intentionally shows this dialog so users can report bugs, which enables me to fix them. I know that these dialogs can be annoying. Do do encounter lots of them? Maybe it would make sense to introduce a preference settings that allows to suppress the dialog, so users who are not interested in testing and writing bug reports are not disturbed by them (although they might be confused by erroneous application behavior).

richardabendroth, 2021-03-15:
Dear Andreas, 
I do encounter quite a lot of error messages. For example randomly when deleting wires or copy-pasting a component (when pasting). Sometimes those error messages lead to an application state where Antares shows the green "circuit saved" toast but in reality stops saving from that point on. When reopening, all editing since the error message is lost. I have been solving this by closing and reopening the Application after an error message was shown. I will try to export a log next time this happens.

flandreas, 2021-03-15:
I know that there is a bug when copy-pasting components from one circuit to another without restarting Antares in between. This has been fixed and the fix will be released with 0.4. Deleting wires or copy-pasting within the same circuit should work, and I would be happy if you could report such cases along with a log file extract.

Yes, restarting the application after an unexpected error has occurred is a good practice, because such errors can disturb the undo/redo command stack, which is involved in maintaining the storable state of the application.

Thank you for your patience!

closed 2021-05-04 by flandreas