flandreas/antares #59: Unexpected Error when starting Simulation

richardabendroth, 2021-03-15:
When trying to start a simulation, an "Unexpected Error" is thrown.  

```2021-03-15 19:11:44,529 ERROR [AWT-EventQueue-0] (SourceFile:73) - error while resolving class 'ch.scorpion.antares.a.g.m' with storableID '44': class ch.scorpion.antares.a.g.m cannot be cast to class ch.scorpion.jabbah.graph.b.D (ch.scorpion.antares.a.g.m and ch.scorpion.jabbah.graph.b.D are in unnamed module of loader 'app')
2021-03-15 19:11:44,535 ERROR [AWT-EventQueue-0] (SourceFile:45) - Error while cloning Storable: class ch.scorpion.antares.a.g.m cannot be cast to class ch.scorpion.jabbah.graph.b.D (ch.scorpion.antares.a.g.m and ch.scorpion.jabbah.graph.b.D are in unnamed module of loader 'app')
2021-03-15 19:11:44,536 ERROR [AWT-EventQueue-0] (SourceFile:26) - Unexpected error: class ch.scorpion.antares.a.g.m cannot be cast to class ch.scorpion.jabbah.graph.b.D (ch.scorpion.antares.a.g.m and ch.scorpion.jabbah.graph.b.D are in unnamed module of loader 'app')
java.lang.ClassCastException: class ch.scorpion.antares.a.g.m cannot be cast to class ch.scorpion.jabbah.graph.b.D (ch.scorpion.antares.a.g.m and ch.scorpion.jabbah.graph.b.D are in unnamed module of loader 'app')
	at ch.scorpion.jabbah.graph.b.d.b.resolve(SourceFile:162)
	at ch.scorpion.jabbah.e.h.a(SourceFile:71)
	at ch.scorpion.jabbah.e.h.a(SourceFile:60)
	at ch.scorpion.jabbah.e.p.a(SourceFile:63)
	at ch.scorpion.jabbah.e.k.a(SourceFile:1061)
	at ch.scorpion.jabbah.e.k.a(SourceFile:27)
	at ch.scorpion.jabbah.graph.m.a(SourceFile:216)
	at ch.scorpion.jabbah.graph.b.g.s.a(SourceFile:194)
	at ch.scorpion.jabbah.graph.b.g.s.a(SourceFile:180)
	at ch.scorpion.jabbah.graph.b.b.a.a(SourceFile:156)
	at ch.scorpion.jabbah.graph.g.aE.a(SourceFile:5214)
	at ch.scorpion.jabbah.graph.g.aU.invoke(SourceFile:1105)
	at ch.scorpion.jabbah.base.b.a.b(SourceFile:73)
	at ch.scorpion.jabbah.d.e.o.b(SourceFile:1348)
	at ch.scorpion.jabbah.graph.g.a.l.invoke(SourceFile:1195)
	at ch.scorpion.jabbah.base.Q.run(SourceFile:22)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
2021-03-15 19:11:51,649 ERROR [AWT-EventQueue-0] (SourceFile:73) - error while resolving class 'ch.scorpion.antares.a.g.m' with storableID '44': class ch.scorpion.antares.a.g.m cannot be cast to class ch.scorpion.jabbah.graph.b.D (ch.scorpion.antares.a.g.m and ch.scorpion.jabbah.graph.b.D are in unnamed module of loader 'app')
2021-03-15 19:11:51,649 ERROR [AWT-EventQueue-0] (SourceFile:45) - Error while cloning Storable: class ch.scorpion.antares.a.g.m cannot be cast to class ch.scorpion.jabbah.graph.b.D (ch.scorpion.antares.a.g.m and ch.scorpion.jabbah.graph.b.D are in unnamed module of loader 'app')
2021-03-15 19:11:51,653 ERROR [AWT-EventQueue-0] (SourceFile:26) - Unexpected error: class ch.scorpion.antares.a.g.m cannot be cast to class ch.scorpion.jabbah.graph.b.D (ch.scorpion.antares.a.g.m and ch.scorpion.jabbah.graph.b.D are in unnamed module of loader 'app')
java.lang.ClassCastException: class ch.scorpion.antares.a.g.m cannot be cast to class ch.scorpion.jabbah.graph.b.D (ch.scorpion.antares.a.g.m and ch.scorpion.jabbah.graph.b.D are in unnamed module of loader 'app')
	at ch.scorpion.jabbah.graph.b.d.b.resolve(SourceFile:162)
	at ch.scorpion.jabbah.e.h.a(SourceFile:71)
	at ch.scorpion.jabbah.e.h.a(SourceFile:60)
	at ch.scorpion.jabbah.e.p.a(SourceFile:63)
	at ch.scorpion.jabbah.e.k.a(SourceFile:1061)
	at ch.scorpion.jabbah.e.k.a(SourceFile:27)
	at ch.scorpion.jabbah.graph.m.a(SourceFile:216)
	at ch.scorpion.jabbah.graph.b.g.s.a(SourceFile:194)
	at ch.scorpion.jabbah.graph.b.g.s.a(SourceFile:180)
	at ch.scorpion.jabbah.graph.b.b.a.a(SourceFile:156)
	at ch.scorpion.jabbah.graph.g.aE.a(SourceFile:5214)
	at ch.scorpion.jabbah.graph.g.aU.invoke(SourceFile:1105)
	at ch.scorpion.jabbah.base.b.a.b(SourceFile:73)
	at ch.scorpion.jabbah.d.e.o.b(SourceFile:1348)
	at ch.scorpion.jabbah.graph.g.a.l.invoke(SourceFile:1195)
	at ch.scorpion.jabbah.base.Q.run(SourceFile:22)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
2021-03-15 19:12:20,458  INFO [AWT-EventQueue-0] (SourceFile:23) - zipping Antares.log

flandreas, 2021-03-16:
I was not able to reproduce the problem. May I ask you to export your project and attach it to this issue? This would allow me to analyze the bug.

richardabendroth, 2021-03-18:
Cannot reproduce it either. After restarting Antares, the simulation runs perfectly. Next time I will try to export it right away.

flandreas, 2021-03-26:
I have produced the same error scenario with one of my test circuits. I will now investigate this..

flandreas, 2021-03-27:
The bug can be reproduced by the following usage scenario:

1. Create a new circuit "A" to be used as part, add some inputs and outputs, design the symbol, save.
2. Create a new circuit "B", add part "A", connect its inputs and outputs with components in "B", save.
4. Start simulation of "B" => Unexpected error.
5. Restart Antares, open "B", start simulation => OK.

It seems what's causing the bug are multiple instances of the library/project objects being around, where the project object instance used for cloning "A" when dragging it into "B" is not the same instance that is used when saving "B" in the first place, leading to unresolvable component references when building the deep simulation graph. When restarting Antares, these library/project instances are loaded from the one and only persistent representation and are therefore in sync.

Implement an automated integration test for this scenario and cleanup management of library/project objects.

flandreas, 2021-03-27:
Fixed. Will soon be released with 0.4.1. The fix should eliminate some of the unexpected errors occurring with 0.4.0.

richardabendroth, 2021-03-29:
Nice! Unbelievable you have been able to reproduce it 😮 👍

closed 2021-04-01 by flandreas