flandreas/antares #65: General Error while opening circuit

richardabendroth, 2021-03-23:
After the following Error occurred after Copy-Pasting a set of circuit elements within the same circuit, Antares refuses to open this circuit with "General Error while opening circuit". 

```Version: 0.4.0
java.lang.RuntimeException: java.lang.NullPointerException
	at ch.scorpion.jabbah.graph.view.graph.a.a(SourceFile:123)
	at ch.scorpion.jabbah.graph.view.graph.a.a(SourceFile:143)
	at ch.scorpion.jabbah.edit.a.n.c(SourceFile:137)
	at ch.scorpion.jabbah.edit.a.x.a(SourceFile:47)
	at ch.scorpion.jabbah.base.l.actionPerformed(SourceFile:65)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Actions.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.SwingUtilities.notifyAction(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBinding(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processKeyBinding(Unknown Source)
	at java.desktop/javax.swing.KeyboardManager.fireBinding(Unknown Source)
	at java.desktop/javax.swing.KeyboardManager.fireKeyboardAction(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBindingsForAllComponents(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBindings(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.KeyboardFocusManager.redispatchEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
Caused by: java.lang.NullPointerException
	at ch.scorpion.jabbah.graph.c.d.b.resolve(SourceFile:182)
	at ch.scorpion.jabbah.e.j.a(SourceFile:71)
	at ch.scorpion.jabbah.e.j.a(SourceFile:60)
	at ch.scorpion.jabbah.e.r.a(SourceFile:63)
	at ch.scorpion.jabbah.graph.view.graph.a.a(SourceFile:95)
	... 53 more
```

Restarting Antares does not solve the issue. Is there a way to fix the corrupted project file?

flandreas, 2021-03-23:
Can you send me your project file? I could the try to fix the corrupted file. And if you can remember which components you copy/pasted, please tell me, then I can try to reproduce and fix the cause of the problem. BTW: Are you using release 0.3.1 or 0.4?

richardabendroth, 2021-03-23:
I am using 0.4.0. It is much more stable than 0.3.1. 0.3.1 used to ignore all changes after some errors when still claiming to save. Version 0.4.0 seems to corrupts files that are saved after an error occurred. 
It was a bunch of components including a subcircuit symbol. 
I can send you the project file. Where does Antares store the files? I have made an export but I can't locate the Antares Project Directory. Will the export be sufficient? 
Where can I send it to?

flandreas, 2021-03-23:
Antares stores your files in the OS-specific default application data directory. On macOS, this is /Users/$USER/Library/Application Support/Antares. On Windows, this is \Users\$USER\AppData\Roaming\Antares, at least on my test machine. Maybe it's worthwhile to make regular backups of this directory as long as Antares runs into these kind of errors.

When exporting a project with File-> Projects and Button "Export..", Antares asks you where to place the generated .zip file containing the exported project. You can choose whatever location you like. If you don't mind, you could attach the exported .zip file to this issue. I case your project is based on your own library (and not one of the standard libraries), I would also need an export of the library in order to open circuits of your project.

richardabendroth, 2021-03-23:
[cabbf62b-23d3-40d1-b394-b66bcab4b6b4.zip](https://github.com/flandreas/antares/files/6192493/cabbf62b-23d3-40d1-b394-b66bcab4b6b4.zip)
I have zipped up the project file from the AppData/Roaming/Antares/projects folder. Maybe it helps in tracing the bug. You don't need to fix it as I can either try to remove some nodes from the subcircuit XML or I will recreate the broken subcircuit. The affected circuit is 'New Circuit'. The other circuits can be opened without any problems. The components affected should be on the very bottom of the circuit if I remember correctly. 
However, the error handling for opening corrupted projects seems like it could use some fault tolerance. Maybe you could allow corrupted circuits to be opened ignoring the faulty components after showing a warning indicating the issues? The user could then decide to save without the corrupted elements or close the project. 
Regarding your question, if I am using a custom Library: I am using the Standard Library.

flandreas, 2021-03-28:
I've fixed the corrupt circuit file "New circuit" by removing a wire that contained invalid references to components. No big deal, that was quickly done. I am able to open the circuit now. Nice project, by the way!

Please find attached the fixed circuit file. You can unzip the file and copy the *.cir file it into the projects directory of your Antares data director, overwriting the corrupt one.

I will now try to find out why the circuit got corrupt.

[919923a7-6c79-489c-a640-e87de781ee78.cir.zip](https://github.com/flandreas/antares/files/6216922/919923a7-6c79-489c-a640-e87de781ee78.cir.zip)

flandreas, 2021-03-28:
Bug fixed. It occurred when copy/pasting only some wires of a more complex net. When saving the resulting circuit, it could not be opened any more due to dangling component references on the net model layer. Also, as suggested by you, reading circuits from persistent storage is now more fault tolerant, although without explicitly informing the user if inconsistencies have been detected, but writing error LOG entries instead.

richardabendroth, 2021-03-29:
Wow thanks a lot! 🚀 🤗