flandreas/antares #68: Feature Request: Open Online Library to share Components

richardabendroth, 2021-03-29:
It would be very neat in the (far) future to be able to share components with other Antares users. This way one could share and make use of i.e. 74xx ICs or other common circuitry. I imagine that in the short time I'm using Antares I've been recreating a lot of components other users have already created for themselves. Likewise, I would like to share some useful common components I have created. An online library with functionality to contribute and subscribe to components would be a killer feature for Antares.

flandreas, 2021-04-01:
Thank you for your input. Yes, absolutely, such a feature would be great, and I envision it to be an essential part of the upcoming web version of Antares, which will contain "social" and community features like showcasing, sharing and liking projects and circuits.

Please note that once bug #66 is fixed and released with 0.4.1, you will be able to create your own library, put the circuits you like to share in this library, export it as ZIP and give it to others, who can then import that ZIP as a library into their Antares installation and use it as foundation of their own projects. Of course, this is still a quite coarse-grained kind of sharing, and I hope the web version will get rid of the necessity to deal with export files, perhaps even allowing to include circuits that others published by just using an URL.

flandreas, 2021-07-06:
I've removed the "Web" tag, because I've restarted to think about this feature and possibilities for making it available before the web version is up and running. Maybe I'll start with importing/exporting individual circuits (including their dependents), and then perhaps a public GitHub projects where people can publish their components using pull requests.

flandreas, 2021-07-08:
As a first step, Antares 0.9.0 will support exporting individual circuits (including all non-standard circuits it depends on) as a "bundle" into a ZIP file. Other users can import this bundle into their own project or custom library.

A place for uploading/downloading bundles will follow (e.g. a public GitHub project). As a first step, it won't support "automatic online sharing"; the users will have to manually download and import the bundles.

Kitick, 2021-07-19:
Awesome, It's a good start at least!
The start to the Antares community, sharing and learning from others as well as from the built in stuff

flandreas, 2021-07-31:
@Kitick @richardabendroth The sharing page is now up and running. Still quite empty though. The first thing I am going to commit by myself is the first version of the 74xx DIL library, once I've implemented #215.

closed 2021-07-31 by flandreas