flandreas/antares #74: Missing special chars in Explorer Properties of "..."

richardabendroth, 2021-05-12:
When highlighting a component with a name containing special chars like "(" and ")" in the Circuit , the name of the component in the Exporer sidebar in the Label 'Properties of "Component Name"' under the pallette tree appears without the special chars. 

Steps to reproduce: 
- Create a Circuit with the name "Circuit (Variant)" and save it
- Create a Symbol for the Component and save it 
- Create a new Circuit and place the "Circuit (Variant)" component 
- Highlight the "Circuit (Variant)" component in the Circuit 
- Check the Label 'Properties of ...' to the left

flandreas, 2021-05-13:
Thank you for pointing to out.

flandreas, 2021-05-14:
The algorithm for creating negated texts using the exclamation character did remove parenthesis even if no negation was asked for. So this problem could also occur in circuit ports, not only in the title of the property panel. Fixed for 0.5.1.

closed 2021-05-16 by flandreas