flandreas/antares #75: Wires used for busses

Kitick, 2021-05-13:
I've been working on building a mini 8-bit computer, and I've run into an issue that stopes the project, wires cant be wired commonly, like a bus. I know this is intentional, to have the wires be directional always, so Input to Output, and not like to like.

However, I'd like to suggest just having wires be common in general, so you can connect anything to anything. This is to have the wires act more how real ones do. I've noticed this becomes a problem as well with making a RAM chip, trying to tie all the outputs to one wire, but it wont let me. This is a little frustrating because it means I have to make a merge chip, to do something that should be done by connecting wires together.

If anything, perhaps have a way to 'force' a connection so that way you can connect signals that you know will work fine. That way, especially with bi-directional chips, I can create a bus that actually can connect to things. Also to other bus transceivers, those cant actually connect to each others wire, you have to connect it into the chip. but that only gives you two of them connected together.

flandreas, 2021-05-13:
Thank you for your feedback.

The rules for creating wires are as follows:

- From input to output
- From output (or wire using ALT) to input **or to another output, if its value can be "undefined"** (which is the case for tri-state outputs)

In the current version, you cannot create a wire from a port to an already existing wire, but this is more like a missing feature, and you can always work around this by creating a junction on the existing wire and connect to the port.

As you mentioned, this rules are intentional to prevent beginners from creating illegal circuits, but they should not be confusing, and they should certainly not stop your project.

You're saying that you can't connect bus transceivers to each other. I assume that you want to connect the "A" or "B" ports of two bus transceivers to the same wire. You do this by creating a junction on the existing wire (using the ALT key), which is already connected to a port of the first transceiver, and drag it to the port of the second transceiver.

Or perhaps I misunderstand you? Can you please describe in more detail what you were trying to that stopped your project, perhaps by posting a screenshot of your circuit that points out the wire you're not able to connect?

Kitick, 2021-05-13:
Alright that makes sense, and yes I knew about the creating a junction from an already existing wire to the port. Though I would agree having the ability to just draw from a port and have it create a junction on a wire would be nice quality of life. It was just confusing for a second

However I did look more into my issue, I was trying to connect other IO devices, like the keyboard interface or a dipswitch to manually put info on the bus for testing, but that's what it wasn't letting me do. When I create the junction, it just doesn't connect to it. But, if I used another bus receiver, then it works, but it just means I have to manually control that receiver, which I guess is fine. I was also having the problem with the receiver to receiver, but I think that was just me not doing the splitting quite right.

Anyway, I got it to connect now, however If I could still suggest maybe having a setting, or a mode in the program that interprets the wires differently? So like if you say that your a beginner, it sets the wires to behave as they do now, but if you say professional, then it could make them common? I only still say this because I came form using other circuit programs where that's how the wires worked, and it was a little more intuitive that way. However all the other things that Antares gives made it a much better choice overall, so keep it up!

flandreas, 2021-05-15:
The reason why you weren't able to connect a junction to a keyboard or a DIP switch is because you targeted output ports, which is not allowed. I will check if it is reasonable to relax this restriction by checking whether there is already another output port connected to the existing wire. If not, connecting from this wire to an output port could be allowed.

Regarding the new feature "New wire from port to existing wire", I have just opened the new issue #83 that covers this. I hope I can implement this for the next minor release.

Regarding your suggestion for a "advanced connection setting", I'm still not sure if there exists any situation (except the one described above) that would not lead to an illegal circuit, and hence produce conflicting net signals during simulation. I will think about it...

flandreas, 2021-06-02:
I'm closing this with implementation of #134 for 0.7.0 and #83 already being possible.

closed 2021-06-24 by flandreas