flandreas/antares #76: Appplication is crashing randomly

richardabendroth, 2021-05-13:
The still crashes randomly when working on a circuit. It is too random to figure out what caused the exception. Things I tink might cause it: 
- Sometimes when deleting several wires really quickly one after the other
- Sometimes when Copy/Pasting either a standard lib component or a user component
- Sometimes when connecting wires 
- When a lose wire end looks like it is connected to a component but isn't and you start the simulation 

Version: 0.5.0
java.lang.NullPointerException
	at ch.scorpion.jabbah.graph.view.b.ad.invoke(SourceFile:1068)
	at ch.scorpion.jabbah.base.i.a.c(SourceFile:199)
	at ch.scorpion.jabbah.base.i.d.a(SourceFile:99)
	at ch.scorpion.jabbah.c.t.g(SourceFile:131)
	at ch.scorpion.jabbah.edit.g.J.b(SourceFile:80)
	at ch.scorpion.jabbah.edit.d.j.b(SourceFile:207)
	at ch.scorpion.jabbah.c.g.D.keyReleased(SourceFile:298)
	at java.desktop/java.awt.Component.processKeyEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.KeyboardFocusManager.redispatchEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

flandreas, 2021-05-17:
Thank you, Richard. I know that these bugs are annoying, and I've now decided that release 0.6 will primarily focus on stability

 I will review the critical code sections and assess their robustness, and I will do more stress testing myself. I'm thankful if you keep sending me stack traces like the one above, which e.g. showed me that the code for user input processing doesn't always properly distinguish between key events and mouse events, i.e. tries to interpret a mouse event when there was actually a key event received, which can result in NPEs.

flandreas, 2021-05-27:
A lot of bugs that might be connected with this issues have been fixed with 0.6.0. This doesn't mean that all are fixed, but I nonetheless set this issue to "done". Please open new issues for 0.6.0 if similar bugs should still occur.

closed 2021-05-27 by flandreas