flandreas/antares #78: Faulty wire/component connections lead to loss of everything created afterwards

richardabendroth, 2021-05-13:
When there is a faulty wire/component connection that is not noticed by the user, every change done after the fault occurred is lost when saving and leaving the circuit. When the user reopens the circuit, the user only sees the open-ended wire (white circle connector) and modifications done prior to the fault ('prior' as in: the order of the circuit's XML nodes). It would be good to be warned before saving or at least to be able to see faulty wire connections.

flandreas, 2021-05-14:
Reproducible. Thank you for reporting this!

flandreas, 2021-05-15:
The fatal root cause of this bug were inconsistencies between the state of the undo/redo system and currently open circuit. If undoing a command (or rollback the current command transaction, as it is the case when aborting a reconnect operation) leads to replaying all commands since the last snapshot, a new circuit instance is created, but this new instance was not rejoined with the currently open circuit. When saving after executing additional commands, and old state was saved.

This bug could manifest in all sorts of situations related with undoing/redoing or opening and saving circuits.

The "faulty connections" now don't occur any more. If the user clicks on the ports and releases the mouse as the same location (thereby aborting reconnect), an info message is displayed saying "Reconnect has been aborted". The info message code was already there for a long time, but didn't show up as a consequence of the bug described above.

The problem with accidental reconnects when the user only wants to move a small component will be handled in issue #77.

closed 2021-05-16 by flandreas