flandreas/antares #79: Feature Suggestion: .circ library import

richardabendroth, 2021-05-13:
Would it be possible to make an import functionality for logic chip libraries like the 7400 logic family library for Logism (https://github.com/r0the/logi7400)? It would save quite some time to just import ready-made ic circuits. Even though it's a lot of fun to recreate them using Antares.

flandreas, 2021-05-15:
Well, in principle yes, as long as the requirement is meant in a similar way like requirement #68.

When it comes to importing library files of other tools such as Logism, I suppose that the required effort to parse such files and to map their model structures to Antares structures would be huge. In the current project phase, I would rather invest my time in stabilizing the existing functionality and the development of the new web frontend.

In addition to that, there is also a conceptual gap. As far as I understand, those 7400 logic family libraries all use DIL symbols. While I see the benefit for building circuits that are based on DIL components, especially if you want to build your circuit finally in real hardware, I think that Antares goes more in the opposite direction. Rather than being a tool for preparing circuits for real hardware (DIL, VHDL, Verilog etc.), Antares is an educational tool and tries to do everything to make circuits more understandable, and DIL packages make them IMHO less understandable.

Again, this is my current understanding, and it might change in the future. But I definitely would love to see how users of the future Antares Web version share their Antares libraries.

richardabendroth, 2021-05-19:
It's not really about using Antares to create a Schematic or program an FPGA. It's more about reusing logic from the 7400 logic IC's without rebuilding it from the respective datasheet for each one without knowing someone else would also be able to use it. I think the proposed feature from #68 hits my suggestion better as I don't really like the user experience (and also the optics) of Logism-evolution. Also, Logism has no feature to actually look into the built-in components when simulating which adds very little to the learning experience.

flandreas, 2021-05-19:
I share your opinion. Would you then agree to close this issue and keep #68 for the future "community shared libraries" feature?

richardabendroth, 2021-05-19:
Absolutely 👍

closed 2021-05-20 by flandreas