flandreas/antares #82: Simulation stopped working altogether

Kitick, 2021-05-13:
Back to working on a little mini 8 bit computer project, I set up a small setup with a keyboard input and RAM, just to test the bus transfers, and a basic control logic, and the simulation is just refusing to play anymore. Correction actually, it plays for a little bit, then gets to a point and then freezes.

I have no idea if its a wired connection I did, or if its something else. No simulation speed setting changes this. The clock doesn't obsolete, and switches don't push values.

Here's a image of my setup; forgive any wired wirings, its just for testing proposes for now.

[image]
[image]

flandreas, 2021-05-14:
Could you please start Antares once more and run the simulation until the problem occurs, and then export the log file using menu "File -> Export Log" and attach the exported log file to this issue? And if you don't mind, may I ask to you to also export your project using "File -> Projects -> Export" and attach it as well so I can analyze what's going on in your circuit?

Kitick, 2021-05-14:
Alright, at least what I can read from this its a null pointer, so I added something incorrectly?

Also I'm not able to open projects, its greyed out, even though I have a project just to store everything. That does remind me, I should make my computer its own project, rather than a circuit. But for now ill leave it as is.

Heres the log file though: [Antares.log](https://github.com/flandreas/antares/files/6476665/Antares.log)

[image]

flandreas, 2021-05-14:
Thank you for the log file.

No, you didn't do anything incorrectly, this is surely a bug in Antares.

The "Projects.." menu item is disabled because the simulation is still running. Please stop the simulation (or restart Antares, if the "Start/Stop" button isn't responsive any more), and try again to export the project. It would tremendously help me to find the cause of the problem.

Kitick, 2021-05-14:
Ah silly me, didn't even think to try that!

Anyway, here it is: [Personal.zip](https://github.com/flandreas/antares/files/6476793/Personal.zip)

flandreas, 2021-05-14:
Thank you very much for providing the project.

I figured out that the source of the problem is Antares' implementation of the RAM component. It throws an exception if input A is "undefined (blue)" while CLK is asserted. This is cleary a bug and will be fixed with the next release. The proper behavior would probably be to make output D also undefined in this case. Thank you again for reporting this.

In the meantime, you could temporarily change your circuit so that CS of the RAM is only asserted if A is defined.

closed 2021-05-16 by flandreas