flandreas/antares #84: Bus Transceivers Outputting Wrong Values

Kitick, 2021-05-15:
Hello again, back to my mini computer project, I just connected up the bus, and re arranges some of the wiring for the RAM chip to not throw the wired undefined error, and apparently, the bus transceivers don't push the right values.

I put the letter 'a' on the keyboard buffer, and it outputs 61, which is the correct hex representation, And it pushes it to the transceiver. And then enabling the keyboard push switch, the transceiver pushes that value. But then out of nowhere, it just decides to push 11 hex to the bus. And that's how it stabilizes.

I've looked into its sub circuit, and I have no idea where the 11 is coming from, it just seams to get it and push it. If I could get some help that would be wonderful!

Here's my current project file: [Personal.zip](https://github.com/flandreas/antares/files/6488353/Personal.zip)

flandreas, 2021-05-16:
Thanks for reporting this. At first glance, this seems to be a bug in the new bi-directional signal propagation algorithm. I am currently analyzing it and revert to you as soon as I have a solution.

flandreas, 2021-05-16:
Thanks to your circuit, I've found a bug in the new bi-direction splitting algorithm and was able to fix it. Your circuit now seems to work. At least I could enter characters and forward them to the terminal. I didn't check the memory operations tough.
I will package a new bugfix release 0.5.1 as soon as possible. Stay tuned...

[image]

Kitick, 2021-05-16:
Awesome, thank you! Can't wait for 5.1!

closed 2021-05-16 by flandreas