flandreas/antares #85: Suggestion for the prorogation delay timings

Kitick, 2021-05-16:
So I as I know, the way the prorogation delay works, is the components wait their specified timeout, and then read the value of their pins when the timer says to. Now I actually had a problem like this for a coding project, and I feel the solution could help here.

So essentially, you just have the component have an array of all the inputs at each tick period, with the length of the array the prorogation delay. And every new tick the values shift down the array. And the current value of the pin at any instant, is the end of the array. Or beginning, depending on if you want it to shift up or down.

This would allow circuits that are very time sensitive to work in Antares, such as a monostable circuit. That only works because of the propagation delay in the and gate. But if the gate only reads it at its timeout, then the not will have already switched, and the circuit wont pulse the length of the not gates delay. It would also help discrepancies in other circuits where they may have specific values at a specific time in the clock period, but can only work because of the real components keeping the values.

Another upside is you could potentially set a different delay for each pin. Now normally, this is not needed, so setting a global delay is perfect. But in some circuits, its noted in their datasheet that different pins have different timings, because of each one being a unique circuit. So that could be something useful.

The only down side to this approach is memory, and processing time. Having to store all the arrays can get intensive, but from what I found in my testing, if you organize the arrays in the object structure of the component efficiently, it turns out not to hinder performance almost at all. And a much of arrays will add a bit to memory usage, but not too much. I feel its worth it for the accuracy and timing advantages it gives. You could always have it as a setting, so the user can choose to use this or not, depending if they want performance or accuracy.

Let me know your thoughts, be much appreciated!

flandreas, 2021-05-16:
I'm not sure if I understand your idea entirely.

First of all, there is no global tick in Antares. Simulation in Antares is event based, meaning that whenever a signal arrives at a component, the component asks the scheduler to call it back for calculation after "prop delay nanoseconds". If another signal arrives during that waiting period, the component will again ask the scheduler for scheduling, thereby remaining responsive even while waiting. But it's true that a component won't calculate new output signals if input signal changes occur during the component's "propagation wait time".

An example of a monostable circuit is the "Pulse Generator" component in Antares' "Logical Gates" folder. Do you see scenarios in which this circuit does not work as expected with the current simulation algorithm, and could therefore profit from a more elaborate algorithm like the one you outlined?

flandreas, 2021-05-16:
Now that I am still thinking about your proposal, I remember that one of the first versions of Antares' simulation engine stored a pins current value in the simulation scheduler's requests, and the component used that value (i.e. the value the pin had at the time when the signal changed) for calculation when the propagation delay was over. I suppose this model is somehow analogous to your "array model". It turned out that I wasn't able to make sequential circuits like flip-flops work correctly that way, and I finally dropped it. But part of the required code (remembering the old values) is still around in Antares.

I once was also wondering whether Antares should let you specify "hold times" on ports, just like component data sheets specify that e.g. and AND gate input must hold its value for a least say 10 ns in order to be effective, and throw a simulation error if the input value changes prior to that time. That would certainly not allow you to finer-timed circuits, but it would help to detect subtle timing design errors.

Kitick, 2021-05-16:
Yeah, also poor wording on my part, I know its event based, I just meant global in reference to the component, being a global setting to all its pins. Not the whole program, sorry for that. Also yes I know that there is a built in pulse generator, that's essentially a pre made monostable, but what if when learning the crude way to make them, with a 'not' gate and an 'and' gate, you have to have that timing accuracy.

And I can see how its a little confusing, so essentially when a new value is detected by the pin, that new value gets pushed to an array, and then is read whatever amount of nanoseconds later by the circuit. this would be for say, if you have a gate with a 10 ns delay, and you inputted values to output a 0 1 0 1 really fast, then that would output that 0 1 0 1 just 10 ns later.

But thinking on it more myself, if you were to change the values of the input so fast, that the transistor didn't switch, would it eventually, like the delay, or fail to switch altogether? Mabey a better approach kind of how you were saying the value needs to be held for a specific amount of time to be valid, then after that has some delay before its output is stable. So then, if you rapid fired inputs to a gate, it would take the propagation delay time to update the last inputted value. So a 0 1 0 1, would just output a 1 10ns later. I feel like that would be more realistic.

To your point though, I feel like throwing an error if the value changes would restrict a lot of circuits, because kind of like with an adder for example, lots of gates rapid fire changes, as carry's update, and the sum just stabilizes after a specific amount of time. I feel like doing something like I said in the paragraph above would be better.

If anything, what you could do is if a new value is detected, the delay restarts. And so its like the hold time that you said, it takes that amount of time before the new input is stable. Because how you have it now, lets say you have a really crappy gate. And it takes 1000 ns to propagate. If lets say, you updated the value, then had a delay for say 950 ns to update the value at the last second, the gate would update correctly 50 ns later. When it should restart that 1000 ns to be stable, if that's the specification.

Let me know your thoughts, my original idea may not be right for this, because of the event based now that I really think about it, but it could be modified to work.

flandreas, 2021-05-19:
The "built-in" pulse generator is actually not so build-in, but really made of an AND and NOTs. Just double-click on "Pulse Generator" and you see its internals. That's exactly one of the things that I want Antares to differ from similar tools: As little built-in, opaque components as possible.

The current implementation uses 3 NOT gates each with 10 ns delay, and the AND gate has the 20 ns delay, resulting in a pulse with of 10 ns. But you as user are free to change the propagation delays of each component.

To make the discussion a little bit more concrete: Do you have a particular circuit in mind which you couldn't simulate in Antares now, and which would work with the "array model"?

If we can agree that there is no global tick in Antares, I could imagine that since you are a software developer as well, you might have been inspired by something like "message queues" that are commonly used in software development to integrate independent software components, each working at their own pace. And this is a wonderful thing.

I would just fear that such an "implicit queue" within every port of Antares components would make Antares circuits be further away from real hardware components, and the probability that your Antares circuit, which would work fine when simulated with Antares, would then terribly fail when built in real hardware, because real components don't include such a "queuing feature", and therefore specify required "signal hold times".

Kitick, 2021-05-19:
Alright I see, interesting, when I went to build one before it didn't work, and I see yeah it does, huh, interesting. It may have been fixed in a release and I just haven't looked back on it, Ill take that as my bad.

But alright, so I just want to rephrase what I was saying, because I totally agree, Antares should be more how a real circuit would work. That's what sparked my other discussion with making the wires just signal relays of sorts, and let the user just put anything to anything, and honestly, think about it this way: if it does result in an illegal circuit, then have the simulation light the wire red, and say hey, you cant do this because this reason, instead of just not letting them do it, and wondering how to loophole it.

But on track with this one, so how I think of it now, is how a real transistor in a gate would operate. If you had a NOT gate, and brought its input high, the transistor would switch off, grounding the output. Turing it off. What I was saying is, when you do that, and then before the gate fully switches, and you switch the input back to high. what happens? Nothing right, because the gate never fully switched on the high state, so it goes back to low.

So that was why I changed my idea, so that, instead of the component reading the values after an input update, then waiting a timeout, it has a hold time as you were saying. So the value only changes to a state, if the value is held for that amount of time. The array model only works if there was a global update tick that synced everything, but the event based method it doesn't work, so that's why I'm modifying it to fit that. And as you said as well, the goal is to have to be as realistic to real circuits as possible, so if you were to build it with real chips that has similar specs, then it should work as you expect.

Unless I'm still not thinking about it the right way still.