flandreas/antares #87: Graphical glitch with text when starting simulation

Kitick, 2021-05-16:
When you have text on the board like I have in a project, if you start the simulation when the text is not in view under the screen, it shows the text. And it stays their even after stopping. all until you restart the simulation with all text visible

Just wanted to point it out.

[image]
[image]

flandreas, 2021-05-16:
Thank you! Yes, that's an old item on my "hidden" TODO list, glad you made it official now :-) Will soon look into it.

Kitick, 2021-05-16:
glad I could help!

closed 2021-05-27 by flandreas