flandreas/antares #88: Suggestion, possibly have wire colors?

Kitick, 2021-05-16:
I just noticed in my project how messy wires can get very quickly, but then I though, wouldn't be neat if you could color the wires? So then you can keep types of signals organized. And how I'm thinking of it, is if you set the color to yellow lets say, you could have the 0 state, or off just be a really dark or even transparent yellow, by changing the luminous value in a HSL color scheme. And convert to RGB if necessary. And the 1 state is the full color / opacity.

Undefined signals could actually be the ones that are transparent, with 0 state being just darkened. Conflicting signals could flash red, as to not conflict with red wires.

I think it could be a nice thing to have as a way to organize signals click control signals or clock signals as one color. Plus it might add some color to an otherwise grey environment. Let me know your thoughts!

flandreas, 2021-05-16:
Interesting idea! And surprisingly, you'd like to add more color, while I recently sometimes was thinking that Antares gets kind of "over-colored", especially while running the simulation.

It would probably be quite easy to let the user set a color on wires just like he can already now set colors on components like AND gates, as long as the colors are only used in editing mode. As you mentioned, mixing the modeled colors with signal colors during simulation would probably require some more analysis.

BTW: Are you aware of the "Style" property of wires? You can use it to let Antares draw imported wires like "Address Bus" or "Data Bus" in a wider, block-styled manner. Perhaps you can use this as a workaround until wire colors are available?

Kitick, 2021-05-16:
Interesting, for me, I'm used to being able to color my circuits based on what does what, so it makes it easier to manage and organize, and can make it look cleaner. But I can see with the over coloring, but honestly, I think its fine. Even if the user was able to change the colors, if they felt like it was too much they could always tone it back

I would want it to also apply during simulation, because that's when the organization that it could bring would help with debugging, but I understand that would be more work.

And yeah, I saw the style option, and never really got into that too much, I should look into it. But really, changing the widths doesn't make a whole lot of sense, maybe for specific instances, but I like the idea of the bigger wires represent the bundles.

Also just wanted to say, the wire bundles are a fantastic feature and every other circuit program should have something like it! Makes stuff that have a lot of bits that are all the same thing like data much cleaner.

flandreas, 2021-05-20:
Due to bug #95, you can't choose the "Style" property for wires, because in 0.5.1. you can't edit wire properties at all. That's why you probably thought I was talking about wires with larger bits width. My bad. I was really talking about block style wires like used for buses here:
[image]
Check it out after #95 has been fixed with release 0.5.2.

Kitick, 2021-07-19:
@flandreas ah alright, cool, I saw this a while ao and never responded :p
Well thats still a great feature on its own, what I was going for was more of, remeber Ben Eaters computer? How all the signels are colored based on what they do? That's more the reason I was wanting colors, because I got used to coloring them. Makes knowing whats what with BIG circuits nicer to look at and debug

flandreas, 2021-07-21:
Okay, I think I can easily color the wires in edit mode, but for simulation mode, I'm not yet sure how to resolve the resulting clashed with the default signal colors (dark green, bright green, blue). I know you've suggested to try applying alpha-blended coloring during simulation.

Maybe we can do this feature in two steps. First, colored wires in edit mode, and release this. After having collected experiences with that, think about applying the colors to simulation as well.

flandreas, 2021-07-21:
@Kitick Implemented, though only for edit mode, not yet for simulation mode. Also added a couple of more predefined colors to complete to color circle (brown, violet, turquoise).

Be reluctant in using brown (orange) and red as wire colors, because they are used by the editor itself (orange for selection, red for "error state").

closed 2021-07-30 by flandreas