flandreas/antares #100: After editing component A used in component B, application refuses to open circuit of component B with 'General Error'

richardabendroth, 2021-05-19:
After editing component A used in component B, the application refuses to open the circuit of component B by throwing a 'General Error'. 

2021-05-19 21:35:17,144  INFO [main] (SourceFile:110) - Starting Antares version 0.5.1
2021-05-19 21:35:17,146  INFO [main] (SourceFile:111) - Using user data dictionary C:\Users\ra\AppData\Roaming\Antares
2021-05-19 21:35:19,330  INFO [AWT-EventQueue-0] (SourceFile:92) - Open 'Computer'
2021-05-19 21:35:31,541 ERROR [AWT-EventQueue-0] (SourceFile:162) - Couldn't resolve Port 1 of Vertice ch.scorpion.jabbah.graph.b.g.r with storableID 246
2021-05-19 21:35:31,733 ERROR [AWT-EventQueue-0] (SourceFile:73) - error while resolving class 'ch.scorpion.antares.view.net.DigitalEdgeView' with storableID '269': Collection contains no element matching the predicate.
2021-05-19 21:35:31,733 ERROR [AWT-EventQueue-0] (SourceFile:45) - Error while loading MetaGraph 318e3148-bbea-4b0c-a94e-5f6f030b78fa: Collection contains no element matching the predicate.
2021-05-19 21:35:49,463  INFO [AWT-EventQueue-0] (SourceFile:23) - zipping Antares.log

richardabendroth, 2021-05-19:
If I recall correctly, I have deleted a port from the symbol and placed it again from the ports pallette onto the same spot in the symbol. The Component has been used in the corrupted circuit and had that port connected.

flandreas, 2021-05-21:
Oh yes, this is such an important scenario, and at the same time a typical example of a developer in the early phase of development saying "I know that this is missing, but it won't bite myself, and I'll implement it before the first release", and then you don't do it at all, until users stumble over it.

Thank you for reporting this :-)

flandreas, 2021-05-21:
Fixed. Antares 0.6.0 will handle these situations and draw the stale wire in error state, like so:
[image]

richardabendroth, 2021-05-21:
Nice solution! 👍

closed 2021-05-27 by flandreas