flandreas/antares #101: How to use the Oscilloscope?

richardabendroth, 2021-05-19:
How can the Oscilloscope be used? The option View > Oscilloscope seems to be permanently disabled.

flandreas, 2021-05-19:
I'm afraid, the oscilloscope is still under heavy construction and not yet publicly available. The menu is only enabled in developer mode. If you are courageous, you can start Antares with "-developer" cmd line option, but make sure you have a backup of your project directory, cause the way Antares currently stores oscilloscope configurations is very pre-beta.

richardabendroth, 2021-05-19:
All right! Thank you! I'll check it out. 😊

richardabendroth, 2021-05-22:
I've tried to use it but only the blue Oscilloscope box appears on the bottom of the circuit. If I click 'Add row' in Edit Mode, it throws an error. If I do it in Simulation Mode it just does nothing. I would really like to play with it and help in finding the bugs 😊 How can I add a pin or something as an osci row?

flandreas, 2021-05-23:
It runs in the development environment, but I can reproduce the error in the released version. Seems to be another Java/obfuscation problem.

I will see if I can make it as least runnable with the next release, but I wouldn't spend too much time on it, because there are unsolved conceptual problem with the oscilloscope, in particular the question of how oscilloscope configurations should be stored. It think that stabilizing the core functionality is currently more important.

richardabendroth, 2021-05-23:
Absolutely. No rush here. I was just curious and eager to try 😊

flandreas, 2021-05-27:
When I tried to quick-fix the exceptions, I actually found out that they only occur if you open the oscilloscope in one of your own circuits. Since you start Antares as a developer, you can currently use it only on circuits that are owned by the developer. So try it out in one of the standard library circuits.

I've just tried it successfully in an installed Antares 0.5.2.

Disclaimer: Changing and saving standard library circuit will throw an exception at you, because by intention they can only be changed in the development environment.

closed 2021-05-27 by flandreas