flandreas/antares #102: How do you import a project

Kitick, 2021-05-19:
So I got the updated project file you sent, but im not sure how to import it, I saw the import button in the projects menu, but then i select import, and go to the folder, (already extracted) it just opens it, with no file to click on.

And of course if you just import the zip it crashes.

flandreas, 2021-05-19:
Make sure you have a backup of your project directory. Then, in the "Projects" dialog:

1. Delete your project "Computer" with "Delete"
2. "Import" the zip file directly (no extracting)

I tried this successfully on my machine before I've sent you the file. The crash happens only if you don't delete your project first. I will register this as a bug.

flandreas, 2021-05-19:
Bug: Antares should not throw an Unexpected Error when importing a library that already exists. It should rather ask the user if it's ok to replace the existing project.

closed 2021-05-27 by flandreas