flandreas/antares #103: How can the simulation cache be cleared?

richardabendroth, 2021-05-19:
Simulations seem to keep some sort of state when running and interacting with (e.g. press toggle switch) them, then stopping the simulation and restarting it. Is there a way to really restart the simulation without opening another circuit and reopening the circuit that held the state? 
Example: 
1. Open Circuit
2. Start Simulation 
[image]
3. Toggle S to 1 
4. Toggle S to 0 
5. Stop (not pause!) simulation 
6. Start simulation
[image]
7. Stop simulation 
8. Open another circuit
9. Re-open original circuit 
10. Start simulation 
[image]

richardabendroth, 2021-05-19:
The problem seems to be with the Probe Component: 

[image]

flandreas, 2021-05-20:
Thank you for pointing this out, great analysis!

Simulation state is distributed among the components and primarily kept in the model layer (primarily ports and wires). There is central code that resets these model states when simulation is started, but the views must be updated accordingly, which can differ depending on the component type. In case of the Probe, it was its view part that wasn't properly updated.

closed 2021-05-20 by flandreas