flandreas/antares #104: Some other Feature Suggestions

Kitick, 2021-05-19:
I have a couple suggestions I wanted to just though out to get your opinion on. Also when I mention the transistor, I'm referencing the transistors that are used in the gates, not the transistor component in Antares.

Also I understand that these are kind of out there, and understand if its too complex or would require to much work.

1. What if there were presets for the delay on gates and other chips to set the transistor type. So for example, the LS and HC for the 74 family is around 10 ns gate delay. so that could be the preset, clicking a drop down of the transited type could set the delay to 10. and there could simply be a custom one that lets you set it manually to whatever.

2. That also gives me an idea, what if there's a power draw stat? So that, you can monitor how much power your circuit would roughly draw in real life. this goes back to the transistor type preset, that also determines the estimated power draw, as well has the gate delay. You can simply look online for all the different types of ones you can use that all have different power draws, delays, voltage levels, etc.

3. This one is definitely out there and is probably much for what it is, but what if you had the wires and components carry a voltage data with it. so, like, instead of a digital 0 or 1, it will say 5V, or 0V. whatever voltage the circuit is using or components are outputting. And you can couple that with the transistor types, and have circuits that may be undervolted or overvolted, and teach new people the importance of making sure the gates used match the power specs of the circuit. Undervolted components may just not switch at all, as they will in a real circuit, and overvolting can maybe throw a simulation error, because well, you killed the component by overvolting.

With all that, especially 3, there would have to be mode settings, like digital to just say, 0 or 1, or voltage to have it show you the voltages. Or maybe, even just have the wire tell you the voltage, and next to it say the interpreted digital value, based on the connected components. Anyway, all this is kind of complex, just want to hear you opinion when you have a minuet.

flandreas, 2021-05-20:
Thank you for your suggestions! I think I understand quite well what you are looking for.

My vision for Antares on this large feature topic is something I'm referring to as "hierarchical systems". Back in the days when I studied computer sciences, I came across the book "Structured Computer Organisation" by Andrew S. Tanenbaum, a book that inspired me a lot and that gave me many ideas that went into the design of Antares.

In that respect, Antares currently operates only on the "digital level". I was thinking a lot about introducing the "analog" level below the "digital level", where the student could open an AND gate and see an analog circuit made of transistors that are effectively simulated in terms of current, voltage and resistance. Now, this level would of course require an entirely different simulation engine based on solving linear equations derived from Ohm's and Kirchhoff's laws, and, when it comes to active elements like transistors, even non-linear stuff.

So, rather than trying to impose too much analog properties onto the digital level, the vision for Antares is more like introducing an explicit analog level beneath the digital level. However, this would require a huge effort, and Antares' roadmap tells me that the web frontend has currently higher priority.

The same applies for introducing the next levels above the digital level, namely the "microprogramming level" and  the "conventional machine" level. Although Antares has already quite distinctive features for explaining what is going on in a microcomputer circuit (signal flow, active controls in symbols etc.), I'm envisioning a level that allows you to  add more abstract objects like "Program counter", "Accumulator", "Heap Memory" or "Stack Memory", and link these objects with the corresponding components in your circuit on the digital level.

But again, although I would love to do this right now, stabilization of the existing features and the new web frontend have currently higher priority.

richardabendroth, 2021-05-21:
I want to add that analog simulations tend to be very, very slow for large circuitry. The huge advantage of Antares is its incredible simulation speed. I personally prefer simulation speed over analog details. For simulating analog circuitry I use [Falstad](https://www.falstad.com/circuit/). It does very accurate analog simulations but becomes incredibly slow for 30+ components, depending on the components' electrical characteristics over time. In Antares there is already a very nice simulation for the gate mnemonics (View > Gate Mnemonics) for learners which, teach the understanding of the underlying concepts.

Kitick, 2021-05-22:
That's true, and I agree, But that's why I said that maybe having two separate states, where one is analog, and one digital, so that if you want to move the abstraction layer, then you can work with just digital, and Antares can make shortcuts to have it simulate faster. Or, go into analog if you want to know what's going on. But as of now, it being on the digital level is honestly perfectly fine, especially for me, since I'm more in the line of making simulated computers and SOCs.

However the gate mnemonics actually kind of do what I was saying, at least for a visual representation, so that's at least a plus.

And also I'd agree, fixing bugs is always more important than adding more features.

flandreas, 2021-05-23:
Although performance is not the main requirement for Antares (explainability is, and there are much faster simulators out there), it should not be neglected too much.

Thats why there is the concept of simulation speed and "deep simulation vs. shallow simulation". If there once would also be an analog layer beneath the digital layer, an AND gate could have its inner analog circuit, and it would also have an expression / a script defining its function. Depending on the simulation mode or the current simulation speed, the inner analog circuit would be simulated or the faster expression would be used. This would allow you to truly build a computer from scratch (starting with silicon you've figged out of the rocks ;-), and, when you're done, run it as fast as possible by using all configured execution scripts.

Thats also the way the "Tanenbaum Computer" on the examples page works. If you run it with "deep simulation", it is 50 times or so slower that when being run with "shallow simulation".

closed 2021-06-24 by flandreas