flandreas/antares #93: Some combination of deleting wires can cause a junction to still just exist

Kitick, 2021-05-19:
I was deleting some wires to reconnect them differently, and somehow I got it to leave the junction, and I cant get rid of it. I can move it, and it even lights up as if it was powered; but its just annoying and I cant delete it. Simply says some elements cannot be deleted. I also managed to do it twice.

There in the top right corner of the terminal

[image]

flandreas, 2021-05-19:
I you don't mind, you can send me your project ZIP and I can quickly remove the dangling junctions and send you back the corrected project.

I assume you don't remember the sequence of selections and deletions that lead to this situation? It would help to reproduce and fix the bug.

Kitick, 2021-05-19:
No problem, and so what I can guess, it happens when I'm doing stuff with multiple wires, that have a junction. But I'm not quite sure exactly what did it.

Also just a side note, have you read my response to the propagation timings suggestion? I know your probably really busy so its fine if you haven't.

But here's the current project: [Computer.zip](https://github.com/flandreas/antares/files/6506651/Computer.zip)

flandreas, 2021-05-19:
Here is your fixed project: [Computer.zip](https://github.com/flandreas/antares/files/6506799/Computer.zip). BTW: Your circuits are growing impressively :-)

No sorry, your propagation timing suggestion is still on my todo list. I'm currently on a deep-dive in fixing critical bugs in Antares, but I will answer as soon as I can...

richardabendroth, 2021-05-24:
Have been able to reproduce. Can also be reproduced by copy/pasting some components that have wires with junctions attached to them. When selecting the components via mouse with a select rect that are already connected to wires with junctions, make sure to only select the components and the junctions but **not** the wires.  Then copy and paste and you are left with some floating junctions that can be moved but not deleted. 
[image]

flandreas, 2021-05-26:
The issue with copy/paste has been fixed. By only deleting wires, I was not able to reproduce the problem, at least with the current 0.6.0-dev version. Let's leave it this way and see if the problem still occurs in the next version.

flandreas, 2021-05-27:
Implemented support for deleting unconnected junctions as to help users to clean up their circuits if the problem should still occur. But please, keep on reporting if the problem persists.

closed 2021-05-27 by flandreas