flandreas/antares #94: Moving Small Circles in Symbol View does not work

richardabendroth, 2021-05-19:
In Symbol View, small circles (i.e. for inverted inputs/outputs) cannot be moved since the circle shape does not show the move cursor. It only shows the other cursors. Additionally, those cursors are offset quite a lot towards the right and bottom of the shape. 

Platform: Windows 10

flandreas, 2021-05-19:
Okay, now I see your point (besides the fact that I recommend to use property "Logic" for inverted ports).

The size of the 8 handles to be used for resizing objects does not change with the zoom factor. This is by intention. However, the sensitivity area of the handles used to decide which mouse cursor to show does not correctly reflect the size of the handles. So this is a bug.

richardabendroth, 2021-05-19:
I think I am not having that option:
[image]

flandreas, 2021-05-19:
Oh no, what a pity! This is a bug, silently swallowed by Antares. Sorry for that. This gives reason for a release 0.5.2 soon.

richardabendroth, 2021-05-19:
Nice! Having worked with Antares and seeing the cleverness you put into it, I should have suspected there was a more elaborated way than mine of creating circles by hand. 🙈

flandreas, 2021-05-20:
Bug "Property not shown" fixed by #95 in release 0.5.2. The "Handles" problem will be targeted in 0.6.0.

flandreas, 2021-05-22:
Fixed the sensitivity area of handles.

closed 2021-05-27 by flandreas