flandreas/antares #95: How to make diagonal wires like in the SR Latch

richardabendroth, 2021-05-19:
Is there a possibility to make diagonal wires like the one's shown in the SR Latch circuit?

flandreas, 2021-05-19:
Select the wire and change the property "Layout" to "Node". When you afterwards select the wire and hover over it, you will receive handles that allow you to move the wire's existing corners. It is not yet possible to add new corners like with the polyline, you can just move the existing corners.

richardabendroth, 2021-05-19:
Where can I change the property of a wire? When I select a wire, the component properties box below the component explorer stays empty.

flandreas, 2021-05-19:
Same problem as in #94. Seems to occur only in releases. Possibly a problem with reflection and Java obfuscation. Release 0.5.2.

closed 2021-05-20 by flandreas