flandreas/antares #105: Error when clicking "Open Subcircuit" in component context menu in circuit

richardabendroth, 2021-05-20:
Version: 0.5.1
ch.scorpion.jabbah.base.c.a: Graph contains a cycle: New Circuit
	at ch.scorpion.jabbah.base.a.l.a(SourceFile:63)
	at ch.scorpion.jabbah.base.a.l.a(SourceFile:71)
	at ch.scorpion.jabbah.base.a.l.a(SourceFile:71)
	at ch.scorpion.jabbah.base.a.l.a(SourceFile:34)
	at ch.scorpion.jabbah.e.j.a(SourceFile:65)
	at ch.scorpion.jabbah.graph.view.b.dT.a(SourceFile:1025)
	at ch.scorpion.jabbah.graph.view.b.dT.a(SourceFile:1021)
	at ch.scorpion.jabbah.e.r.a(SourceFile:63)
	at ch.scorpion.jabbah.e.m.a(SourceFile:30)
	at ch.scorpion.jabbah.e.a.a(SourceFile:43)
	at ch.scorpion.jabbah.graph.view.graph.GraphViewImpl.cloneForExistingModel(SourceFile:173)
	at ch.scorpion.jabbah.graph.view.vertice.SubGraphVerticeViewImpl.createSubGraphView(SourceFile:356)
	at ch.scorpion.jabbah.graph.g.ab.a(SourceFile:183)
	at ch.scorpion.jabbah.graph.g.W.invoke(SourceFile:1240)
	at ch.scorpion.jabbah.graph.g.M.a(SourceFile:194)
	at ch.scorpion.jabbah.graph.g.M.a(SourceFile:3239)
	at ch.scorpion.jabbah.graph.g.U.invoke(SourceFile:1215)
	at ch.scorpion.jabbah.base.Q.run(SourceFile:22)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

flandreas, 2021-05-21:
Does this problem only occur when using the context menu, or also when double clicking on the symbol?

The stack trace says that the inner circuit contains a cycle. Do you remember anything special you did with the subcircuit, like exchanging components?

richardabendroth, 2021-05-21:
Sadly I don't. There was nothing special that I noticed. I couldn't reproduce it but it might be connected to the bug that sometimes the option is disabled for no obvious reasons and sometimes clicking the option just does nothing.

flandreas, 2021-06-01:
Most probably the same as #119 and #130

closed 2021-06-10 by flandreas