flandreas/antares #111: Error when selecting multiple components and deleting them

richardabendroth, 2021-05-21:
Version: 0.5.2
java.lang.NullPointerException
	at ch.scorpion.jabbah.graph.view.b.dU.a(SourceFile:51)
	at ch.scorpion.jabbah.graph.view.b.dU.e(SourceFile:108)
	at ch.scorpion.jabbah.graph.view.a.p.a(SourceFile:140)
	at ch.scorpion.jabbah.edit.c.b.a(SourceFile:127)
	at ch.scorpion.jabbah.graph.view.a.c.a(SourceFile:1113)
	at ch.scorpion.jabbah.edit.a.m.a(SourceFile:36)
	at ch.scorpion.jabbah.edit.a.i.a(SourceFile:41)
	at ch.scorpion.jabbah.base.l.actionPerformed(SourceFile:65)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Actions.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.SwingUtilities.notifyAction(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBinding(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processKeyBinding(Unknown Source)
	at java.desktop/javax.swing.KeyboardManager.fireBinding(Unknown Source)
	at java.desktop/javax.swing.KeyboardManager.fireKeyboardAction(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBindingsForAllComponents(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBindings(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.KeyboardFocusManager.redispatchEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

richardabendroth, 2021-05-21:
Same happens when doing the same to pack of wires. So I guess it's somehow connected to removing nodes from the tree and then have pointers to nowhere in other components?

flandreas, 2021-05-24:
Yes. The bug was caused by old code that tried to keep "unconnected" information used for "reconnecting" when undoing actions. Undoing is now based on replaying snapshots, and this code now been removed (already done with another refactoring for 0.6.0). So, set to "done".

closed 2021-05-27 by flandreas