flandreas/antares #112: Feature Suggestion: Expandable muti-bit inputs/outputs

richardabendroth, 2021-05-21:
It would be a handy feature to have Circuits with i.e. 4-Bit inputs to feature an option at the inputs to automatically expand those to a combiner (or splitter for outputs respectively) so the user does not need to place a splitter/combiner manually. This makes a lot of sense for ICs where you would like to group the D_IN and D_OUT so the learner has it easier to understand the meaning of the pins while still being practical for connecting those to individual wires when needed.

flandreas, 2021-05-21:
Can you elaborate a bit more? How would the component look like if you choose this option for a particular input? Would there a combiner hanging right at the pin of this input? Or would the pin show a popup when the user draws a wire to this input, asking him which of the 4 bits he wants to connect? Would this then graphically look like 4 wires be connected to a single pin?

richardabendroth, 2021-05-21:
I like the 2nd idea a lot. Like you said, the individual wires could just end up in the bigger pin. Or the pin could have a little square on it that denotes that. When clicking or hovering the square (or whatever looks good) would then show an overlay visualizing which wire goes to which of the x pins. This could also be used for editing. So when pulling a wire onto the square (with mouse still down) the overlay opens and the user can connect to the individual pins within the pin bundle.

richardabendroth, 2021-05-21:
Maybe like this?

[image]
[image]
[image]
[image]
[image]

flandreas, 2021-05-24:
Thank you for your detailed design proposal, that's impressive! However, I still feel a little uncomfortable with this feature. First, it would be quite hard to implement, as it breaks a couple of existing invariants, such as a pin has at most 1 connected wire on the view layer. I also doubt that we could find a solution for how to layout multiple wires arriving at the same pin that doesn't confuse the user.

Second, I looked around to see if other tools contain such a feature, and it couldn't find one. In terms of requirement engineering, I would now ask you for a concrete circuit were you really miss this feature. I went over the "Tanenbaum Example" and couldn't find a case where this feature would have been beneficial.

Maybe your requirement arise from the fact that the Antares standard library contains a lot of components with multi-bit pins? Maybe this is wrong, and multi-bit pins should only be used when it is quite sure that the user will bring a wire with the same width? And if a component library mainly contains components with lots of single-bit pins, we should have more convenient features to easily concentrate single-bit-pins right at the pins of a component? And this is a feature that I DID see in other tools..

Kitick, 2021-05-27:
For me, what I understand is that you want it to basically auto create splitters or mergers so that you don't have to have extra components in the circuit. But in my opinion, I'd think that it would be pretty difficult to implement because of how the current way the wires and pins work. But also to the fact that in my circuits, I've only ever needed splitters or whatnot for I/O. And honestly, set the prob delay to 0, and its like its not there. So I agree with flanderas that while it may be an interesting feature, I don't see many use cases and/or the need for it over just simply adding a extra splitter every so often.

I also agree that a better solution would be to just have more components that have more bit width options. I'd say at least in powers of two up to 8. So it would have 1, 2, 4, and 8 bit width variants. Or, just have 1 and 4. and if you need 2, cascade two 1s, and 8 is two 4s. (However 8 is a nice number for data transmission)

closed 2021-06-24 by flandreas