flandreas/antares #113: Feature Suggestion: Negated input on Gates

richardabendroth, 2021-05-21:
It would be very practical to have to option on gates to negate some inputs so the user does not need to add extra inverters for each input. 

[image]

flandreas, 2021-05-21:
Yes, I was once missing this feature too. The only reason why it has not been implemented yet is because it's  cumbersome to implement in the properties view along with the varying number of inputs. I will think about it.

richardabendroth, 2021-05-21:
Also, showing the respective mnemonics will be a hard task to be fit into the limited space available within the gate.

flandreas, 2021-05-22:
Don't you think that a negation will just invert the position of the "micro switch" in the mnemonics, so that the "micro mnemonic circuit" doesn't have to be changed? I haven't thought about it in depth, but at least that would be my hope.

Kitick, 2021-05-22:
I would agree, all you'd need to do is flip the gate. So if in the OR gate they two start with the switch open, then a negated input would start with the gate closed.

But in practice, Negating the inputs of an OR is the same as just using a NAND gate. Unless your only negating some of the inputs, but then you usually have to use a dedicated inverter anyway. Because the rule of inversion, is type and input. So a inverted OR, is a NAND. and AND is NOR. OR is if any are on, NAND is if any are off. AND is if all are on, NOR is if all are off.
XOR and XNOR are special however, because they compute parity.

You can prove that by looking at the mnemonic for a OR and a NAND. There the same, just inverted. and a AND and NOR. So really, inverted inputs are already done. You just use its inverse gate.

Kitick, 2021-05-22:
However for other components, like the bus transceiver specifically, it would be nice to specify if its active LOW or HIGH, because my control signals are set HIGH, so I have to have a bunch of inverters inorder to use the transceiver. So being able to set an inverted input, or heck, a monostable input, wither rising or falling per pin, would be a nice time saver. The triangle symbol for clocks is an example.

flandreas, 2021-05-23:
I understand what you mean. This touches on an important restriction you have to face after you've decided that a library should consist of modeled circuits and not of build-in, programmed ones: There are no parameters the user can choose from when using a library component, namely "bit width", "active LOW/HIGH" etc.

The are other tools that try to solve this problem by a concept called "generic circuits". When you design a reusable circuit, you can also define "generic parameters", e.g. "Bit Width". When the circuit is reused, the user can then define a value for "Bit Width". There are various strategies of how the subcircuit implements such a "generic parameter". The simplest one is probably that the subcircuit contains a circuit variant for each possible parameter value. The there are other solutions, such as recursion, and so on. Very advanced topic..

For your problem with the bus transceiver, it's probably not necessary to got that far. To summarize: If you add a subcircuit to a circuit, you would like to be able to "implicitly invert" each input of the subcircuit using properties in the property view of the subcircuit? This would add an implicit NOT to the input, and would also add an "inverter circle" to the input?

Kitick, 2021-05-23:
That's sort of where I'm thinking, is that you would be able to add as a property of the subcircuit if the input is to be active low or high, and/or if it's edge sensitive. But, that's less realistic though, because some of what you do with designing circuits is working around the pre built parameters of your chips. But really, in a program, it might be nice to have that ability to do change them.

But really, I don't think it would require any special algorithm. Just have each component take the input, and compute the parameter form it's attribute. And if you view the subcircuit, it just adds a NOT gate or monostable circuit right after the circuit input. And/or removing it.

flandreas, 2021-05-24:
Yes, that makes sense. And when it comes to adding the NOT gate when viewing the subcircuit (and thereby possibly running into  layout problems), we could also think about just adding an "invert circle" to the inout component in the subcircuit.

flandreas, 2021-05-24:
BTW, this feature will also give reason to rework the truth table displayed as a tooltip above the gate. These truth tables are currently static and restricted to two inputs. I will make them dynamic, respect the new configurable inversion, and also support more than two inputs, perhaps up to 4 or so, avoiding them to become too large.

My mind implodes immediately and refuses any further service if I ask it to calculate the output of a 3 input NAND gate with two inverted inputs ;-)

richardabendroth, 2021-06-02:
You could then also add an IMPLY and NIMPLY gate to the standard library to have the complete set of basic textbook gates.

flandreas, 2021-06-12:
Implemented. The truth tables are now dynamic and reflect the input negations. Just like the mnemonics also do. Even on the OR shapes, the negation circles don't look as terrible as I feared.

https://user-images.githubusercontent.com/24633141/121782044-f9736b80-cba7-11eb-9136-d5e61436a0ee.mov

Only the tooltips now aren't accurate any more if there are negated inputs. But I leave it that way for the moment. Maybe we can improve that once the DeMorgan equivalents are implemented.

richardabendroth, 2021-06-12:
Woooow this looks soooo amazing! 🤩

closed 2021-06-24 by flandreas