flandreas/antares #114: Feature Suggestion: Continuous Overline on Port Names

richardabendroth, 2021-05-21:
When using the 'inverted' syntax in Port Names (e.g. "!(LOAD A)") the Overline (bar) on the produced label is not continuous. It would be great if the line would be uninterrupted and fitting to the char width of the respective char. 

[image]

flandreas, 2021-05-21:
Yeah, that looks ugly indeed. The feature is currently implemented using special unicode characters that produce the overline, and as far as I understand, the width of the overline depends on the font actually being used. I will look around to see if I can find a better solution for this problem.

Kitick, 2021-05-23:
Well actually one solution is to use HTML. Antares's UI is coded in HTML correct? So really all you'd need to do is set the style of the label to add the overline. I don't remember the specific code however I do remember it exists.

flandreas, 2021-05-24:
Well, Antares' UI is coded in Java using the Swing framework, but some parts like the multiline text is drawn using a particular Swing component that can draw HTML text. But as far as I know, this HTML rendering component does not have great CSS support, if any at all. And the port labels are currently not drawn using HTML.

I was also thinking about a compromise solution: If the entire text is inverted, then the overline could be drawn programmatically just by "draw line", because the size box of the entire text is known at that moment.

flandreas, 2021-05-26:
Implemented the compromise solution: Draw the overline programmatically if the entire text is negated. Currently only applied to pin labels and other component labels, but that's the most important usage anyway.

[image]

The solution can be extended later, especially when it comes to other text decorations or things like subscripts.

closed 2021-05-27 by flandreas