flandreas/antares #115: Feature Suggestion: Enlarge Muti-Port Gates to be able to show Mnemonics

richardabendroth, 2021-05-21:
The option to show the Gate Mnemonics is just wonderful! It should be enabled by default for the learners to learn. However, when using gates with more than two inputs the Mnemonics disappear and the gate ports look somehow non-intuitive. 
[image]

It would be nice if the whole gate enlarges (the light-blue background with round corners) so it covers all gate inputs. This would also provide the space necessary to display the mnemonics. 

[image]

flandreas, 2021-05-24:
Glad you like the mnemonics!

Well, the design with the extended line to which the additional ports are attached seems to be widely used, not only in other tools, but also in "manually drawn" circuits. If you google "and gate 8 input", you see that quite a lot. And I think I once saw this also in a standard specification document, but I wasn't able to find it again.

But I admit that it was also easier to implement than the variant with the enlarged rounded body. Maybe it's also a matter of taste. While your enlarged NAND gate look pretty nice, I saw enlarged OR-shaped gates in the web that look terrible.

When it comes to the mnemonics, I'm not sure if it's worth the effort trying to offer them also for gates with more than 2 inputs. After all, it's kind of a "beginner" feature, and beginners tend to not use gates with more than 2 inputs. And most of the mnemonics will anyway be impossible to build for more than 2 inputs, e.g. all the X gates (exlusive OR variants).

Kitick, 2021-05-27:
I think the larger mnemonics would be really cool to have as a visual! If it can be done as you said then I'd agree with it.

Also you said XOR gates would be impossible with more than two, how so?

closed 2021-06-24 by flandreas