flandreas/antares #119: Error when starting simulation after replacing a component into existing wires

richardabendroth, 2021-05-22:
After deleting a JK-FlipFlop and placing a custom FlipFlop with the same footprint at exactly the same position within the circuit wiring (all wire connectors yellow) and starting the simulation the following error occurred: 

```
Version: 0.5.2
ch.scorpion.jabbah.base.c.b: Illegal state: Port already connected
	at ch.scorpion.antares.c.b(SourceFile:3022)
	at ch.scorpion.jabbah.graph.model.port.PortImpl.connectTo(SourceFile:104)
	at ch.scorpion.jabbah.graph.model.d.f.resolve(SourceFile:157)
	at ch.scorpion.jabbah.e.j.a(SourceFile:71)
	at ch.scorpion.jabbah.e.j.a(SourceFile:60)
	at ch.scorpion.jabbah.e.r.a(SourceFile:63)
	at ch.scorpion.jabbah.e.m.a(SourceFile:30)
	at ch.scorpion.jabbah.e.a.a(SourceFile:43)
	at ch.scorpion.jabbah.e.a.a(SourceFile:28)
	at ch.scorpion.jabbah.graph.m.a(SourceFile:228)
	at ch.scorpion.jabbah.graph.model.f.r.a(SourceFile:218)
	at ch.scorpion.jabbah.graph.model.f.r.a(SourceFile:204)
	at ch.scorpion.jabbah.graph.model.b.a.a(SourceFile:148)
	at ch.scorpion.jabbah.graph.view.N.a(SourceFile:1093)
	at ch.scorpion.jabbah.graph.view.P.invoke(SourceFile:1045)
	at ch.scorpion.jabbah.base.b.a.b(SourceFile:73)
	at ch.scorpion.jabbah.d.e.o.b(SourceFile:1344)
	at ch.scorpion.jabbah.graph.a.k.invoke(SourceFile:1086)
	at ch.scorpion.jabbah.base.Q.run(SourceFile:22)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

```

richardabendroth, 2021-05-22:
Same Problem, different Ciruit: 
```
Version: 0.5.2
ch.scorpion.jabbah.base.c.b: Illegal state: Port already connected
	at ch.scorpion.antares.c.b(SourceFile:3022)
	at ch.scorpion.jabbah.graph.model.port.PortImpl.connectTo(SourceFile:104)
	at ch.scorpion.jabbah.graph.model.d.f.resolve(SourceFile:157)
	at ch.scorpion.jabbah.e.j.a(SourceFile:71)
	at ch.scorpion.jabbah.e.j.a(SourceFile:60)
	at ch.scorpion.jabbah.e.r.a(SourceFile:63)
	at ch.scorpion.jabbah.e.m.a(SourceFile:30)
	at ch.scorpion.jabbah.e.a.a(SourceFile:43)
	at ch.scorpion.jabbah.e.a.a(SourceFile:28)
	at ch.scorpion.jabbah.graph.m.a(SourceFile:228)
	at ch.scorpion.jabbah.graph.model.f.r.a(SourceFile:218)
	at ch.scorpion.jabbah.graph.model.f.r.a(SourceFile:204)
	at ch.scorpion.jabbah.graph.model.b.a.a(SourceFile:148)
	at ch.scorpion.jabbah.graph.view.N.a(SourceFile:1093)
	at ch.scorpion.jabbah.graph.view.P.invoke(SourceFile:1045)
	at ch.scorpion.jabbah.base.b.a.b(SourceFile:73)
	at ch.scorpion.jabbah.d.e.o.b(SourceFile:1344)
	at ch.scorpion.jabbah.graph.a.k.invoke(SourceFile:1086)
	at ch.scorpion.jabbah.base.Q.run(SourceFile:22)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

```

flandreas, 2021-05-24:
Cannot reproduce with standard components. Could be a problem with your custom circuit you are moving in.

Are you able to open and simulate the custom flip-flop without problems? In any case, would you mind sending me the custom flip-flop that is causing the problem for analysis?

flandreas, 2021-06-01:
Same as #130

closed 2021-06-10 by flandreas