flandreas/antares #121: Question: Why are all Flipflops high on startup?

richardabendroth, 2021-05-22:
I have a rookie question: 
Why are all Flipflops and Latches high on startup? Is this by definition? I've been implementing some 74xx ICs alongside building some circuitry to learn things but I stumbled upon problems when i.e. implementing a cascadable 4 Bit Binary counter. The counter would start at 1111 and stay there until the first clock impulse since the Flipflops would be high when starting the simulation. 
Also, other components give a high impulse on startup like i.e. the  'Shift Register Parallel-In 8 Bit'. When starting the simulation I can see that Q goes high before stabilizing to low after a brief amount of time. I've now helped myself by either using the Flipflops upside-down or just re-implementing Flipflops with the desired startup value.  
Has anyone an expert's advice on how to do this the right way? I'm afraid I'm making things harder than they should be.

Kitick, 2021-05-22:
It depends on the chips your using. Some specify that the gates start pre loaded, or the output is negated, but mostly with gates derived from the 74 Family, they start cleared. And I noticed that as a problem as well in my 8-Bit computer, where the counter started on 1111, so the first instruction was actually the last. But I just used a different flip flop.

Speaking of the counter though, its a little inconvenient to have the built in counter be 3 bits, instead of 4. Makes scaling to nice powers of two more difficult, and or just wastes pins. I actually needed the counter to be able to be set, so I had to make my own anyway, but just something to comment.

flandreas, 2021-05-23:
That's a somehow arbitrary choice. It is determined by the propagation delays I've chosen when building the latches and flipflops of the standard library.

If you open e.g. the "SR Latch" in the standard library, you see that the upper NOR gate has propagation delay 11, while the lower NOR gate has propagation delay 8. If the choice would be exchanged, the latch would power up Q=0.

It this disturbes you, I can easily change this. Of course, this would mean that all circuits out there start to behave differently from the moment people install the new release. But I think there are not that many user out there yet, and we could still change it.

Should we?

Kitick, 2021-05-23:
For me I'd say so. Since actually thinking on it, most gates say that it's almost random what state the latch starts in, and recommends to have the circuit set the latch to a known initial value before using the data from it.

But, if we were to not use random, I'd rather have the flip flops start cleared. Just makes the most sense. Also side bit, I think all the flip flops listed in the library should have a built in preset and clear functionality, just makes it more usable. And maybe possibly at a set function to the counter? So that it can be used as a program counter. Or would that be more on the lines of just having a program counter IC as you stated in a different discussion?

richardabendroth, 2021-05-23:
Wow, thanks a lot for the quick answers! 😊 From the learner's perspective, I also think it would make sense for me/them to start low. Just imagine someone trying to learn something about counters with a simple circuit and the first thing they need to find out is either how to implement a reset on power-up or re-create the Flipflop circuit with inverse propagation delays and then recreating the counter with that Flipflop (like I did). I would go with changing it t.b.h. 

Regarding the 4 Bit counter: It's quite easy to build one based on the 3 Bit Counter that is provided. This is also a nice learning experience. It might be a feature for later to add counters with more bits, preset, clear, and cascade ability. Keep in mind that there are so many different types of counters: sync, async, sync clear, async clear etc. pp. It would take up a lot of actual development time to add them now. 

Regarding PRE/CLR for the FlipFlops: Most of them have a comparable version that has a PRE/CLR. If one does not need something really special there are sufficient types there for the moment in my opinion. I would volunteer to create some to be added later so @flandreas can focus on the development.

flandreas, 2021-05-24:
Thank you all! So then were are going to change the power-up behavior of the standard library's latches and flip-flops with the next release.

closed 2021-05-27 by flandreas