flandreas/antares #122: Feature Idea: Component that is connected to a Digital I/O USB-Module

richardabendroth, 2021-05-23:
What do you think about having a special component that can be linked to a Digital I/O USB Module so that the inputs/outputs of the component mirror/control the digital I/O's on the USB Module? This way a Simulation in Antares could be directly connected to a breadboard with an led or a PS2 Keyboard or an LCD Display, etc. The learner could then step by step interact with real components and move the designed circuit to a breadboard piece by piece. Is this a silly idea?

Kitick, 2021-05-23:
Wow, that sounds really cool, I actually have a couple USB interfaces. The only problem I could see, is it would probably take a lot of work to get something like that implemented, and validated. But still an interesting idea, and definitely would help with the learning aspect. If you happen to have real components and chips that is.

flandreas, 2021-05-24:
That's an interesting idea, but probably one that is a little bit more down the roadmap.

closed 2026-08-16 by flandreas