flandreas/antares #123: Suggestion about the simulation time

Kitick, 2021-05-24:
So as it is right now, Antares runs the simulation, always advancing to the next available signal update. And this is all well and good, however, I wanted to ask if maybe there was a way to set it to be constant? I mean this in the way that, if the user wanted, time would advanced at a constant simulation rate, and signals would still update at their respective times. There'd just be more blank time in between.

Now I know I must sound crazy, why would I want it to run slower, well I started getting a little disorientated with the simulation in the second speed range, (forgot the name sorry!) and everything kind of happens almost at random. And when I enable the time stamp, it is just jumping everywhere. And that's to be expected, but I just am asking this for a sense of timing, so I can have the sense of really how fast something runs, without that I can see a wire update, and think it happened immediately after something else; when actually it took another 20ns.

In a perfect world there'd be a setting for both. Weather the simulation runs constant, or skip to the next signal update. I feel that there are times in witch both have their pros.

flandreas, 2021-05-24:
That's a legitimate point. Actually, this is kind of already implemented. If you design a simple circuit with one timer connected to an LED, and you set the timer frequency to 1 second, you see that the simulation is effectively slowed down, and the LED gets turned on/off every 1 second. So, when simulating, Antares, tries to make sure that simulation time doesn't run faster than real time.

Additionally, if you don't run the circuit at full speed (speed slider not at right-most position), Antares adds a little wait time to every simulation cycle. The effect you are experiencing probably comes from the fact that this added "slow down wait" time is absolute, but should probably be shorter if the current simulation step requires more work.

But yes, I'm aware of other issues regarding simulation timing (e.g. issue #1), and I agree with your requirement that circuit simulation can be slowed down evenly, and I will work on that after the stabilization phase.

flandreas, 2021-05-24:
Regarding remembering "Display simulation time in state bar": No, this doesn't get stored intentionally, because this setting slows down simulation quite a bit. It's therefore intended to be used only selectively and consciously. Are you missing that strongly?

Kitick, 2021-05-27:
Ah alright, interesting. That's a nice workaround for what I want to do thank you! But it would be nice to have that as just a simulation setting rather than having to add a random LED lol.

But thank you for the input though, I look forward to it! And of course I understand after the bugs are fixed, I agree that's always more important.

richardabendroth, 2021-06-11:
I would like to add that the simulation speed jumps quite a bit when moving the handle from one section to the other. The fastest explore feels a magnitude slower than the slowest observe. In use, everything is just lightning speed. I would love to have something in between the fastest explore and the slowest observe. I wouldn't even care if with explore or observe visuals.
What is kinda missing is some very fast way of pausing (like hitting the space bar after clicking a button) and potentially even reverse playback for some short amount of time.

Kitick, 2021-06-11:
@richardabendroth I totally agree, I actually touched on that in my first box, that the simulation felt Jumpy as you said, and felt hard to keep consistent what was happening when, and also I agree with what you added with a way to make the individual states slower or faster, because the fastest speed is lightning fast, and the others are much slower. Its not as proportional.

flandreas, 2021-06-13:
It is not proportional by intention, because the speed range covers many magnitudes. If it would be proportional, you wouldn't be able to fine-tune the speed at the slow end of the range.

But you are right that the current implementation is not yet optimal; it's a large topic on its own, also related with signal animation and the question whether slowing down should also apply to the execution of hidden sub circuits. Personally, when I really want to see what's going on in a complex circuit, I use "Explore" with signal flow animation, even with single step mode.

But as I've said, there is still a lot of work and thought to do in this area. But it's an important area, because it's one that's supposed to distinguish Antares from other tools.

Kitick, 2021-06-13:
Ah I see, and I totally agree @flandreas the slowest mode that literally shows you the signals travelling down wires and sowing their values, is a really cool visual to watch and learn what's happening; and its one of the reasons I really liked Antares in the first place! Even though it was very buggy when I first started, like version 0.2 or 0.3, I saw the potential. And when I found your git from your website and learned your actively developing it, I was like, yeah this is my kind of program and developer! I just like how your making it with the goal of learning, and explanation, and an emphasis on visuals. Its really awesome and some really nice features like bit width, and gate mnemonics, tunnels, and the other tools make advancing complexity layers simpler, without abstracting really. And its really great, I just wanted to say that. Just keep it up!

Anyway, side from that I think the different modes can be magnitudes faster, but I think you should be able to fine tune the speeds of the simulation within each level. So if the slider is just barley over the second mark, it executes slow. And if its near the other side without going to the third, then its faster. But I can see how you said it, with it not supposed to be proportional on propose.

richardabendroth, 2021-06-14:
I think I might not understand how to use the single-step mode. 🙈 Except for components starting to blink and show their propagation delays I don't see how to advance a single step. The whole thing runs through until the circuit is stable.

flandreas, 2021-06-15:
1. Enable menu Simulation -> Soft Breakpoints.
2. Optional: Use simulation speed "Explore" or larger to fast-forward circuit setup
3. Start the simulation
4. Once the circuit has stablized, press the "Pause" button
5. Optional: Use simulation speed "Observe" to activate signal flow animation
6. Do something in the circuit, e.g. press a switch
7. The "Resume" button should turn orange when the simulation stops at a breakpoint
8. Press "Resume" to proceed to the next breakpoint

All the animations and breakpoints during the circuit setup phase is kind of annoying. I'm thinking about changing the code such that signal flow animations and soft breakpoints are not active before the setup phase has ended, because normally nobody is interested in observing the setup phase.

closed 2021-06-24 by flandreas