flandreas/antares #124: Odd wiring behaviour

richardabendroth, 2021-05-24:
Sometime the auto-router changes wiring down the path with odd results. Sometimes even leaving the user with wires that can't be moved anymore and are somehow broken. 

[image]

[image]

flandreas, 2021-05-26:
Fixed the following scenario that annoyed me for a long time:

[image]
[image]

The wire being split is not kept more stable.

[image]

flandreas, 2021-05-26:
Also improved logic to avoid non-orthogonal layout solutions. That's all I can do for now. More optimizations would have to be done based on reproducible scenarios.

closed 2021-05-27 by flandreas