flandreas/antares #126: Clock as symbol control

flandreas, 2021-05-27:
After every big release (especially after releases with a lot of hard bug fixing like 0.6.0), I need to develop a "recreational fun feature".

This release's fun feature will be to enable clock components to be added to symbols using the "controls" feature. It will show the clock as a turnable knob, featuring the big knob that pops up in circuits when hovering over a clock component.

I need this feature to turn one of my test circuits into a cool example circuit to be placed on the web page. It showcases the powerful possibilities of controls in a great and fun way.

[image]

The circuit consists of a bunch of LEDs that are arranged to form a sine curve, and that are driven by a sequence of shift registers. The push button is used to send a changed signal into the curve, and the clock is used to control the speed in which this signal propagates along the curve.

The project will contain a special "front panel circuit" that solely consist of the LED's, the push button and the timer's knob.

Kitick, 2021-05-27:
Interesting concept, I'm not quite sure I understand it, but it seams really fascinating to be able to have more visuals!

flandreas, 2021-05-27:
There is a feature called "controls", described in the user manual in section "Controls" of https://www.antarescircuit.io/user-manual/english/subcircuits/subcircuits/. It allows you to put some of a circuit's components into the symbol of your custom circuits. This is mainly for components like switches and LEDs, and it allows you to make the symbol active and interactive. This is IMHO one of the most distinguishing features of Antares compared to other tools.

The clock component has a feature called "knob". If you hover the mouse over a clock during simulation, a big knob appears allowing you to change the clock frequency during simulation.

Now, the clock component is not yet a "control" that can be added to a circuit's symbol. Once this is possible, you can build a symbol that allows the user to change the frequency of a clock within the subcircuit.

When I talk about "front panels", I'm referring to the idea that a circuit contains a symbol with all interactive components of that circuit, and then you add a second "front panel circuit" to the project that only contains this subcircuit, meaning that when you open this "front panel circuit", you only see the interactive elements, no other circuit components like NANDs or flip-flops.

This circuit then acts like the "front panel" of your hifi-system, only LEDs, switches and knobs :-)

flandreas, 2021-05-31:
Here's the current state of implementation.

https://user-images.githubusercontent.com/24633141/120184635-7f3f0080-c211-11eb-9eea-343fd5721279.mov

Kitick, 2021-05-31:
Cool! I think I understand it now, and yeah that is a nest feature and visual to have.
Also neat sin wave!

closed 2021-06-24 by flandreas