flandreas/antares #129: Suggestion, add bit width to all gates?

Kitick, 2021-05-27:
This is similar to the feature of adding the width to the NOT gate, but with other gates as well.
So basically, if you have an AND gate, then you could set its bit width to be whatever, and it would preform the AND operation on each bit pair individually. As if you split them off and re combined the outputs.

So for example a 0111 0110 AND 1011 1110 would return a 0011 0110. And so on for any gate type.

Please let me know what you think!

Kitick, 2021-05-27:
One reason this could be useful, is if you want an enable AND gate, but you want to enable 8 Data bits. Instead of having to use 8 And gates, you could just use one.

flandreas, 2021-05-28:
In your example of the "enable AND gate" for 8 Bits, would that mean that pin 1 has 8 bit and the "enable" pin has 1 bit, so it must be possible to set the bit width for every pin individually? Or would all inputs be 8 bit, and the "enable" signal must also be 8 bits, which would be kind of unusual?

Kitick, 2021-05-28:
Well, it would be all 8 bits if that's the width. but what's useful is if you needed an enable like that, you could just take the enable signal, and tie to to all the inputs on a merger, then take that into the gate. Yes its a little work, but its a lot less work than using 8 individual gates with the enable lines tied together.

I actually found the name for this, its called Bitwise operations. And its used in some programming applications with binary specific data. And well, for a circuits program, basically all the data is binary specific.

Hope that clarifies some of it!

Kitick, 2021-05-28:
Or actually, I just thought of this, you could add a component like a replicator, that scaliest or reduces one bit width to another.
Put simply, it would be a component that for example, you could tie one bit as an enable signal, and replicate that to an x bit width constant. That would make the enable use case easier to implement in circuits.

flandreas, 2021-05-28:
Yes, good idea. If all bits have the same width (which is simpler and therefore preferable), you would need something  like you mentioned as "replicator". That would be a useful component anyway. I think it's the same that is called "Bit Extender" in Logisim.

richardabendroth, 2021-05-31:
@Kitick what you want is something like this: 
[image]
The bit mask in M defines the outcome of the operation: 
0001 | A NOR B
0010 | A AND NOT B
0011 | NOT B
0100 | NOT A AND B
0101 | NOT A
0110 | A XOR B
0111 | A NAND B
1000 | A AND B
1001 | A XNOR B
1010 | A
1011 | A OR NOT B
1100 | B
1101 | NOT A OR B
1110 | A OR B
1111 | 1

Kitick, 2021-05-31:
Well sort of, that seams to be a very complicated way to do something with just parallel gates, plus if I'm following what the logic is, you essentially just made a gate computer with multiplexors, what I was looking at was a way to have bulk compute ability with a single gate, and it does it based on bitwise, so each bit individually.

richardabendroth, 2021-05-31:
I see but this is also kind of the way that configurable logic gate ICs work which are in fact something you can buy even though those are kind of unobtainium in a DIP package. Those are quite low in propagation delay as is my suggested circuit. The point I am trying to make is that I think gates should be gates and logic should be logic. Somehow I have the feeling that it's not a good thing for learners to have magic somewhere that can't be obtained as an actual IC when trying to experiment on a breadboard.  Wrapping this circuit in a component is much more explicit than just hiding it in code magic in the basic gate components.
Nevertheless I do think the 'replication' functionality would be quite interesting.

Kitick, 2021-05-31:
I totally agree that nothing should be put behind a magic code block, but technically you can represent what I'm proposing with gates, all it is is the same thing as taking a splitter and splitting the inputs to a bunch of gates in parallel.

So for example, if you wanted to have an enable gate for data input, and let's say the data was 8 bits, then you'd need 8 Dual Input AND gates. One for each data to compare to the enable, and you just tie all the enables together. However what I'm proposing is to have some abstraction in that regard, to where if you put in a 8-bit AND gate, it would perform the AND operation on all the bits in parallel as if they were connected like that.

This goes for any of the standard logic gates.
But I see what you mean, perhaps there could be a nemonic that shows this?

flandreas, 2021-06-01:
Interesting discussion.

@richardabendroth Wouldn't your point about "code magic" also apply to the n-bit inverter we've just implemented with 0.6.0? I kind of understand if users now say "okay, let's have the same for all the gates". Or how is the inverter different from the other logic gates?

richardabendroth, 2021-06-01:
Interesting perspective! I've only looked at it from the input perspective since all other gates have multiple inputs I felt Inverters and Buffers should be no different. But indeed that also changes the outputs. So for my better understanding of the discussion: 
How would this look in different scenarios regarding inputs and outputs?
Would a 4 times AND gate with two inputs have four times a two bit input wire and one 4 bit output wire? 
Would a 1 AND 4 gate have one single input wire, 1 four input wire and 1 four output wires? Or would in the 2nd case the AND gate have 1 four bit input and 1 four bit output wires and a 'side' wire like the tri-state buffer?
How would this be configured in the properties? 'Bit with' and 'Number of inputs' for the 1st scenario but for the 2nd where the bit with is different for A and B? How many different bit withs are possible within one gate like A, B, C, D, ...? What would be the difference to a multiplexer?

Kitick, 2021-06-01:
Alright so here let me explain what I'm proposing, and also why you might have some confusion

So what I'm saying is you can take a logic gate, and set it's number of inputs to be whatever, and in addition, set the but width. The bit width does not change the input or output pins, it simply allows you to do logic operations in parallel on multiple bits without having a bunch of gates.

I think what you were saying was to have the input and output pins be different widths, and thats a little more confusing to operate on, and even to implement. I'm not actually sure what that actually means in terms of logic.

Quick thing, one thing I've learned from my years of computer science, especially with simulation science, is there's two major factors in high level simulators. There's complexity, and usability. There inversely proportional. The more complex, the more useful and practical. However it's less user-friendly and understandable. And vice versa.

So really, we just need to decide how much abstraction we want to include. I come from an area where you learn the little stuff, then abstract and move up levels. That's why all my suggestions have been abstraction suggestions. And some other stuff like wire colors.

richardabendroth, 2021-06-01:
Okay so do you mean something like a 74x00 NOR gate IC or something like a multi-bit AND gate that has an enable bit connected to all and gates (similar to a transceiver)? Or both?

Kitick, 2021-06-01:
Well that's just an example, I don't know any other examples on the top of my head, but like when you need to use bitwise operations in programming for specific operations that are binary heavy, in a circuit program, everything is binary, so it could be a nice quality of life to have.

I guess that IC is a made example of how you could do that with real chips, but again that goes back to what I was saying, we should decide how abstract we want the program to be.

Kind of like the inverter having the multi-bit width, like what's the difference between an inverter and an AND or an OR gate? I feel like if that gets to have multi-bit than the other gates should too.

flandreas, 2021-06-02:
I think it all started because the tri-state buffer. It supported BitWidth right from the start, because this was needed to conveniently build tri-state components connected to wide data busses.

Then came the   observation "if the buffer, then why not the inverter". At least, they look similar. Because other tools also support n-bit gates, I agreed. And then, obviously, came the request "if the inverter, then why not the other gates".

I see the need for them, and I also see the reservation about "black magic" and "you can't show the internals of such a n-bit gate".

If we implement them, users can still decide freely if they want to use them or not. Beginners won't, experts perhaps will. If I was a teacher and was to design a microcomputer to be used in classes, I probably wouldn't, because their functionality is not obvious, and I would rather resort to building sub circuits like "data path selection using ANDs with select input" from scratch, starting with 1-bit components.

Also, they are not too hard to implement. The biggest challenge will be to place the negation circle at inputs of XOR gates, where there is currently not enough space.

richardabendroth, 2021-06-02:
I don't think anyone would mind if the bubbles overlay the bar to be honest. I mean having the functionality is worth a little trade-off regarding the looks.

richardabendroth, 2021-06-02:
I have played with this functionality in Logism and it seems to be very complicated to understand in the beginning. But i think with the much better visuals of Antares (i.e. more explicit visuals for multi-bit inputs/outputs) it will be way easier to understand how it works. The only thing I would like to mention is that the beginner needs to understand that the thing he/she is looking for is 'Number of Inputs' and that 'Bit With' is something for more advanced users. Otherwise it's just super confusing. Is there a way to highlight the 'Beginner Properties' in the property view so beginners that don't know the nuances of the wording don't get confused?

Kitick, 2021-06-02:
I'm not sure actually, but that does give a practical idea, @flandreas There should be a setting in Antares where you can set your experience level, and lets say there's 3 levels, and each one gives more abstractions. Where as Beginner would not let you use bit width, and intermediate does, maybe that's a good way to balance the abstractions?

flandreas, 2021-06-07:
Perhaps yes, although this would probably require a significant amount of work, e.g. you would have to cover the case where the application runs in beginners mode and doesn't show "bit with" properties, but then the user opens a circuit from the standard library where "bit width" is used. Would you then show the property? Probably yes..

richardabendroth, 2021-06-07:
Another suggestion: how about just adding some slight background to the 'beginners' or 'that's probably what you want' properties in the property editor? 
[image]

flandreas, 2021-06-09:
The properties view can also display explanations for every property if you click on the right-most button in its header. Maybe that would help too. Although most properties don't have explanation texts yet.

The discussion reminds me of a suggestion my son made once. He was inspired by a video by Sebastian Lague (https://www.youtube.com/watch?v=I0-izyq6q5s&t=1034s) in which he kind of builds a computer from scratch. The logic gates are just rectangles with the corresponding labels "AND", "XOR" or "NOT". My sun suggested to provide this as an additional symbol style, in addition to the existing "American" and "European" style. Being a teenager, he would call this "Noob mode" :-)

richardabendroth, 2021-06-11:
This is a very cool and simplified way of showing how things work together. Really clean and reduced to the minimum information needed to understand a concept. What I find a bit odd is that the pins don't have any visible description. While this is not a problem for gates where everyone assumes that the side with one pin must be the output, the registers are just a nightmare when you have to figure out that the last pins on the left are WE and CLK. Not even the fact that everything always goes left-in right-out solves that sufficiently. Just looking at a Screenshot will not make you able to understand what's happening. I like the descriptiveness of Antares way more.

Kitick, 2021-06-11:
Interesting yeah I agree, but I see why its called "Noob mode" that gave me a chuckle. And I agree with @richardabendroth I like Antares way of visualizing gates as well, and the "American Style" instead of the "European Style" where its just rectangles with symbols like ~ ^ and &. I like the gates having a shape and giving a visual representation as @richardabendroth said. And that ICs have labeled pins, and while yes real chips don't, you have to consult the datasheet, its better for learning and designing purpose's to have them named right there. And then, you can figure out the logic from a screenshot.

But having an option to switch between the styles could be a useful feature.

flandreas, 2021-06-13:
@Kitick You can switch between the two styles using menu View -> Symbol Style, even without restarting Antares, and even during a simulation run.

Kitick, 2021-06-14:
Ah I see thank you!
Also @flandreas I found another example for this, with the bus transceivers. All you are doing is doing the same operation but on each bit together right? So having this would allow you to build a single bit bus transceiver, and you can change all the gates to the desired bit width, making a multi bit transceiver without needing to build more circuitry.

What do you think?

flandreas, 2021-06-24:
@Kitick Well, I think the examples you mention go more into the direction of "generic circuit" which we've discussed in other issues. The problem is that generic circuits are straight forward if the generic parameters can be directly applied as parameters of circuit components. It gets very hard if these parameter values have an impact of the circuit's graph structure, such as duplicating parts of the circuit n times and connecting the outputs of these parts to the outputs of the surrounding circuit.

Kitick, 2021-07-17:
@flandreas that does make sense, and for that scenario yes i can agree that it would work better there.

I did also think of another use case, lets say in a binary adder, you wanted to selectively invert all the input bits for twos complement, well youd simply run an XOR against each bit. Well, thats the same thing as the enable example I was talking about. Just a way to do the same operation in parallel without a bunch of gates

flandreas, 2021-07-19:
@Kitick Okay, you've convinced me, so let's do it :-)

closed 2021-07-30 by flandreas