flandreas/antares #130: Error when starting simulation

richardabendroth, 2021-05-30:
Hi Andreas, 

the overall stability of the application improved massively with version 0.6.0. Tremendous achievement! 
I have found another bug, which I seem not to be able to reproduce. It occurred when starting a simulation right after placing a user component twice (from the explorer, no copy-paste) and interconnecting it: 

```
Version: 0.6.0
Change property 'Fan Out' of component 33 to '2'
Move component 'Combiner' with ID 33 by mouse
Start creating new EdgeView at PortView
Connect output of SN74HC191 4-Bit Sync Up/Down Counter with input of Combiner
Move component 'Combiner' with ID 33 by mouse
Start creating new EdgeView at PortView
Connect output of SN74HC191 4-Bit Sync Up/Down Counter with input of Combiner
Move segment of EdgeView 35
Move segment of EdgeView 35
Add Component 'Probe' from library/project
Change property 'Bit Width' of component 36 to '8'
Start creating new EdgeView at PortView
Connect input of Probe with output of Combiner
Add Component 'Switch' from library/project
Start creating new EdgeView at PortView
Connect output of Switch with input of SN74HC191 4-Bit Sync Up/Down Counter
Start creating new EdgeView at PortView
Connect output of SN74HC191 4-Bit Sync Up/Down Counter with input of SN74HC191 4-Bit Sync Up/Down Counter
Change property 'Toggle' of component 38 to 'false'
Enter execution mode
ch.scorpion.jabbah.base.c.b: Illegal state: Port already connected
	at ch.scorpion.antares.c.b(SourceFile:3022)
	at ch.scorpion.jabbah.graph.model.port.PortImpl.connectTo(SourceFile:104)
	at ch.scorpion.jabbah.graph.model.d.f.resolve(SourceFile:161)
	at ch.scorpion.jabbah.e.j.a(SourceFile:71)
	at ch.scorpion.jabbah.e.j.a(SourceFile:60)
	at ch.scorpion.jabbah.e.r.a(SourceFile:63)
	at ch.scorpion.jabbah.e.m.a(SourceFile:31)
	at ch.scorpion.jabbah.e.a.a(SourceFile:43)
	at ch.scorpion.jabbah.e.a.a(SourceFile:28)
	at ch.scorpion.jabbah.graph.m.a(SourceFile:228)
	at ch.scorpion.jabbah.graph.model.f.r.a(SourceFile:217)
	at ch.scorpion.jabbah.graph.model.f.r.a(SourceFile:203)
	at ch.scorpion.jabbah.graph.model.b.a.a(SourceFile:147)
	at ch.scorpion.jabbah.graph.view.N.a(SourceFile:1093)
	at ch.scorpion.jabbah.graph.view.P.invoke(SourceFile:1045)
	at ch.scorpion.jabbah.base.b.a.b(SourceFile:73)
	at ch.scorpion.jabbah.d.e.o.b(SourceFile:1344)
	at ch.scorpion.jabbah.graph.a.k.invoke(SourceFile:1089)
	at ch.scorpion.jabbah.base.Q.run(SourceFile:22)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

```

flandreas, 2021-05-31:
Thank you, Richard, I'm glad the progress is also visible to you.
And thank you for reporting this bug. When looking at the stack trace, it seems to have the same cause as #119. I will look into it..

richardabendroth, 2021-05-31:
By the way, I love the new logging so it's visible what has been done right before the error occurred!

flandreas, 2021-06-01:
Quite a nasty bug this was! Unbelievable it could remain undetected for so long.  There were indications though...

closed 2021-06-10 by flandreas