flandreas/antares #131: Feature Suggestion: Rotary Encoder

richardabendroth, 2021-05-31:
It would be nice to have a rotary encoder as an input component as it is not really possible to emulate this using the existing input components except from two switches and a lot of clicking in the right sequence.

flandreas, 2021-05-31:
You mean something like "Joystick" in Logisim? What resolution would you expect for the output signals for x and y? Would 9 positions be enough? (8 deflection positions and 1 center/idle position)?

Or something like a representation of an angle, but then without a center/idle position?

richardabendroth, 2021-05-31:
I mean something like this: 
[image]

While the 'knob' is rotated it gives a constant clock pulse on one output and another clock pulse on the other output that is 90° out of phase compared to the first output. For turning right the phase shift is +90°, for turning left it is -90°. 

I imagine you could utilize something like the clock wheel for user interaction with the switch? But it should only emit pulses when the user is actually 'turning' it; like a momentary button only emits a pulse when the user presses it.  

It could additionally feature a third output that is just a momentary switch ('pressing the knob') but that's just glitter since an additional switch in the circuit would do as well.

flandreas, 2021-06-01:
Okay, I see. It makes me wonder on what kind of project you are working on... :-)

This is quite a specific component, and I am trying to keep the built-in components as general and widely usable as possible. Wouldn't a "Joystick" component à la Logisim also suit your needs? If Antares would have a "Joystick" component, maybe some users of the Logisim family would decide to start using Antares.

If the joysticks "idle middle position" would break your system, we could perhaps have a property for disallowing this middle position?

richardabendroth, 2021-06-01:
I am learning about binary counters and BCD and thought this is an easy way of learning about implementing user hardware interaction. This use case does not necessarily require debouncing so it could be done with digital logic only. 
[image]

flandreas, 2021-06-02:
So would a general joystick component also suffice your needs?

richardabendroth, 2021-06-02:
I honestly don't know. Would it produce the same sequence of phase-shifted pulses?

flandreas, 2021-06-07:
I don't now yet. But perhaps we could have a configurable "signal pattern" on the joystick.

richardabendroth, 2021-06-07:
Sure, why not. I'm currently thinking about just using some gates that produce two phase-shifted signals and package them into a subcomponent like my HEX switch in #136.

richardabendroth, 2021-06-07:
I found an actual component that would come very close to the UI of the existing DIP switch: 
[image]

richardabendroth, 2021-06-09:
I've built a component out of basic Antares Components and it works okay-ish. As mentioned in #136 I'm having problems with the right delays. But it should be able to illustrate what I mean. 
[image]
Here is the circuit: 
[image]

flandreas, 2021-06-10:
Impressive!

flandreas, 2021-06-13:
First joystick implementation.

https://user-images.githubusercontent.com/24633141/121810184-15d2df00-cc60-11eb-978b-36dc952f245c.mov

Produces signal x: Indicates knob's x-coordinate, to be interpreted as an unsigned integer whose value will never be 0. Thus, a value of 1 represents the far left, and the maximum value for the bit width indicates the far right. When the knob is at rest (in the center), the value has the bit pattern 10...00.

Same applies for signal y.

The joystick has a property for choosing "rectangular" or "circular" deflection.

flandreas, 2021-06-13:
And here a run with the oscilloscope:

https://user-images.githubusercontent.com/24633141/121815485-0e6b0000-cc77-11eb-9e7c-ea52374e932a.mov

richardabendroth, 2021-06-14:
Nice! So when I just use bit 0 of each X and Y, I would exactly get the behavior of a rotary encoder! Now I understand! Cool! By the way, the oscilloscope is such a cool feature! Can't wait for it to arrive in userland 😊

flandreas, 2021-06-15:
Yes!

closed 2021-06-24 by flandreas