flandreas/antares #135: Names of Components with enumerated Names

richardabendroth, 2021-05-31:
This is not really a bug nor is it a missing feature, just something I have observed: 
When dragging enumerated components (like inputs or outputs) into a circuit that already contains at least one of those components the symbolic representation of that component contains 1 in its name (like I1 or O1) but as soon as the component is placed it changes its name to the first free number available (e.g. I5 or O2). Shouldn't this be calculated already before the component is placed? 

[image]

Kitick, 2021-05-31:
Id suppose so, but really it shouldn't matter because you should always change your inputs and outputs names to represent what they do. The I1 I2 O1 O2 are more just placeholders.

flandreas, 2021-06-01:
Since Antares desperately tries to make sure that circuit port names are unique, I hope that the above screenshot was captured while you were still dragging the second port into the circuit.

I get your point. When reproducing this issue, I've also noticed that the component being dragged is not selected (orange), which perhaps adds to the confusion.

From a technical point of view, the problem kind of is that with drag'n drop, you don't know where the user is dragging the component to. Although there is currently only one target circuit, this could change in the future. You can also drag a circuit symbol from the project to the library, which results in moving the circuit, and this complicates things even more.

We could change the port name when the port is being dragged over a circuit, and change it again if it is hovering over another circuit. I will check in the code if this could be done easily.

richardabendroth, 2021-06-01:
It's not a problem at all, just something I have noticed. I have done some user testing in the past and know that developers (like I am one as well) appreciate having those small observations recorded when spotted. But I don't want to be intrusive or annoying so please tell me if I shouldn't record them here 😊

flandreas, 2021-06-02:
I appreciate your inputs, even if the should't lead to a software change. Please keep them coming :-9

closed 2026-08-16 by flandreas