flandreas/antares #136: Feature Suggestion: Hex Switch Input with Default Value

richardabendroth, 2021-05-31:
In addition to the DIP switch it would be nice to have a HEX- or n-switch. The output is a multi-bit output much like the DIP switch but the UI is a turn wheel like: 
[image]

The switch should feature a default value as well. Maybe it could also allow to set the number of positions like 4, 10, 16, etc. For 16 it would be neat to feature labels from 0 through F.

Kitick, 2021-05-31:
That's kinda already there, you can have a circuit input, and change the representation to HEX. But are you asking more for the visual part? Because the functionality would be just that.

richardabendroth, 2021-05-31:
Ah interesting! 😊 Didn't see that. Just tried it out. Thank you! 
Nevertheless I still like the idea of the user experience that a rotational switch provides.

flandreas, 2021-06-01:
This issue touches on basic usability and UI design questions. Should a simulator also reproduce features of the "real system being simulated" (in this case real hardware circuits) that are designed based on the restrictions of the real word (in this case the screwdriver slit), or should it take advantage of the possibilities of the simulating system, where the uses has a keyboard he can use to enter hex keys, hence no need to fall back to the "screwdriver world".

I'm not sure what the correct answer is. If an Antares user is going to build the designed circuit in the real word too, then the "screwdriver component" possibly makes sense. If not, then perhaps not, and I would make more sense to add a "representation" property to the DIP switch, allowing the user to directly enter hex character using the keyboard, just like he can do with the circuit inputs.

Kitick, 2021-06-01:
@flandreas
I feel like the circuit input is a definite abstraction layer for the input and outputs of a circuit, but I feel like if you wanted to have an analog version, then that's where stuff like real transistors and and rotor switches and and throwback switches, and all those components would start to matter.

It kind of goes back to what I said in a previous discussion, I don't remember which one it was, but Antares right now is very much a digital circuit program, but it has the potential to have some analog parts to it as well. We just need to figure out how we want to utilize both worlds

flandreas, 2021-06-02:
Yes, that's a similar discussion like in #139 and others. It's about supporting low level details (gate level and even below) vs. supporting abstractions. To me, both aspects are interesting, valuable and important. But when it comes to allocating development time, I slightly prefer abstraction at the moment, because one of the overall goals of Antares is to help explain how computers work by building one yourself - and then program it.

Also, due to the architecture of Antares and its way towards web capabilities, I think Antares has a potential to help explaining those big, abstracted concepts better than other, comparable tools. And that where I'm targeting at with Antares.

Nonetheless, I sometimes dream about supporting analog circuits, and even being able to open a transistor and see electrons and holes flowing through semiconductor regions :-)

Kitick, 2021-06-02:
I think that could be a really fun way to see circuits, seeing it to that kind of level. There was one program I tried out that showed the flow of current, and was really pretty. Had lots of visuals and colors, however the horrible user interface, lack of components, no way to copy paste, make ICs, no way to select multiple components, and even the board you were given was limited in size, and just general lack of usability made it not a good choice for me. And that's actually when I found Antares, and honestly, you've done a great job even so far, and I think with more development, it could become a high level software!

Anyway, I do like the concept of building and programming computers from gates, that's kind of what I've done in my 8 bit project, now moving on to 16 bit and DDR style RAM, but its only really possible with Antares, and not really many others. And trust me, I've tried many softwares I cant even name them all.

So then the verdict for me is this:
Antares should stay in the digital world for now, focus on getting all the core features and bugs ironed out, and have abstractions yes, but each one can be unpacked for a greater understanding. Like with the ICs currently, you can just view it. Stuff like that. And after we get to a point, then we can start the overhaul of a analog variant with voltage levels, real transistor logic, breadboards, rotor switches, resisters, etc.

flandreas, 2021-06-02:
.. and don't forget Antares Web with online interactive text book, homework assignments, and all the cool social sharing and collaboration stuff :-)

richardabendroth, 2021-06-05:
I have to say that I'm kinda lost as I don't get how we got from basically a more intuitive version of the existing multi-bit input in the also existing HEX mode to analog circuitry 😅 Just to be able to understand this for future suggestions: what is the reason again why this feature request is going against Antares' general direction?

flandreas, 2021-06-07:
I try to summarize. The suggested component contains a wheel to be turned by a screw driver. This is useful in real world circuits. If Antares would mainly be a tool for circuit pre-production, this would be a great component. 

The feature request is about setting hex values during simulation by avoiding to use the circuit input component. The DIP switch currently supports binary representation only. Why not add a "Representation" property on the DIP switch and allow the user to enter hex characters using his PC keyboard just like he does on circuit inputs?

richardabendroth, 2021-06-07:
Yes exactly, no screwdrivers, please! It is indeed just an alternative input method for the DIP switch. I would somehow prefer to have a mouse-input functionality rather than typing it with the keyboard. This is much more intuitive. I didn't even know about the keyboard feature until you mentioned it.

richardabendroth, 2021-06-07:
I have found an actual component that has a very similar UI to the existing DIP switch: 
[image]

flandreas, 2021-06-07:
Cool  👍

richardabendroth, 2021-06-09:
So I've built a component that does the job but I'm still having some problems. 
[image]
[image]

- When clicking the arrows the output jumps to a different value before settling on the right one. I don't know how to prevent this. 
- The delay timing is very complex and finding the right delay that makes interaction user-friendly in all simulation speeds seems to be somewhat hard to achieve. In real-time mode, a click has to be super short. Do you have some timing suggestions for me?
- There is no way to implement a default value right now. 

Here is my circuit: 
[image]

The buttons react on 'release' for a single click and when held pressed for a while they increment/decrement continuously until released.

richardabendroth, 2021-06-09:
@flandreas would it be possible to have this (or something similar) as a standard component? Maybe as a variant of the DIP switch? I'm not sure if, when using a circuit, this can be made as glitch-free as the DIP switch. Also, there is no default value possible (yet) for user circuits (see discussion in #152 ).

flandreas, 2021-06-10:
Okay :-)

closed 2026-08-16 by flandreas