flandreas/antares #138: Odd hold behavior of momentary switch

richardabendroth, 2021-05-31:
I'm not sure if this is just a bug or a buggy feature but when a momentary switch (togggle=false) is double-clicked it locks into the 'on' state effectively making it a break switch. Subsequent clicks output 0 until released. While this is a VERY nice feature especially for debugging a circuit it does not work reliably. Sometimes it requires some repeated clicking until the switch accepts the double click. Also, the double click triggers two outputs, 1 0 1 0 or 0 1 0 1, depending on the state it is 'locked' into which made me think this is actually not an intended feature. 

[image]

Kitick, 2021-05-31:
So you found a way to turn the button into a switch? Maybe that could be a feature, if you right click or double click a button, or momentary switch, it toggles it's output.
And just clicking it normally sets it back to normal.

richardabendroth, 2021-05-31:
However, like it is now, repeatedly clicking a momentary switch very fast basically breaks the switch. It should work one way or the other but somehow reliably.
But coming back to your idea: this could be also achieved by having a 'Make/Break' property in the switch. Maybe thats a way to approach the problem. It would at least spare an inverter in between.

flandreas, 2021-06-01:
I haven't checked yet, but that's probably a bug. Switches aren't supposed to respond to double clicks. Note that switches also disable themselves while they wait for their propagation delay to be over. Maybe that's the moment where event handling gets confused. I will check..

flandreas, 2021-06-01:
I've double-clicked as wildly as I could on a non-toggle switch, but failed to reproduce your observation.

However, when you click/hold beside the non-toggle switch, move the mouse onto the switch, and then release the mouse button, the switch locks into "1" state. I will fix that.

flandreas, 2021-06-01:
BTW @Kitick: You seem to be a native english speaker. I've struggled to determine the correct terms that distinguish permanent and non-permanent switches, even by using Google and Leo. In German, a permanent switch is called "Schalter", while a non-permanent switch is called "Taster", at least as far as I am aware.

@richardabendroth is using the term "momentary switch", while the current translation file uses "Push Button" vs. "Switch", and the property for distinguishing them is called "Toggle". Maybe we can improve the consistency of these terms.

So want would in your option be the correct terms here?

Kitick, 2021-06-01:
@flandreas Well, it seams your looking for a button. A switch just toggles it's state when clicked, as you have them now, and a button only is active when you click down. The current monetary switch.

So my vote would be to name them Switch and Button.

I just thought momentary switch was a more systematic name for it, but I did think it was a little odd, but made sense to me at least.

richardabendroth, 2021-06-01:
So property named Type having two options Switch and Button?

richardabendroth, 2021-06-01:
By the way: are we still discussing the property 'Mode' with 'Make' and 'Break' options that can be toggled with a right click like @Kitick suggested?

Kitick, 2021-06-01:
Here's how I understand it

There would two components in the standard library called a switch and a button.
The switch toggles its state when clicked, and a button temporarily is active when you click down as you would expect. I feel like there shouldn't be any property on either it should just be the component itself that does its purpose.

However, we could also have it to where when you right click instead of left click, it will perform the action of the other component. So if you have a switch right clicking it makes it act like a button, and if you have a button right clicking it makes it toggle and stay active. This seems to be a little more complex, but it could be a interesting quality of life.

But we could just have an input device component, that has that property that you're talking about that you can switch it to be either component without having to replace it. I actually like that better, because it means you can switch between switches and buttons without having to delete and replace.

flandreas, 2021-06-02:
Yes, property "Type" with options "Switch" and "Button", instead of the property "Toggle".
Besides that, I don't think we need another new option.

flandreas, 2021-06-02:
Aaah, now I see: It's all about right clicks! I didn't get that until now. I've only tested with left clicks...

richardabendroth, 2021-06-02:
Yeah I just reproduced it. With right clicks it can be reproduced consistently but if you click too fast with the left mouse button (e.g. fiddling with a counter) it also happens.

richardabendroth, 2021-06-02:
I found out how to reproduce it: 
- Place a Switch with Toggle=false 
- Start the Simulation
- Left-Click outside of the Switch and hold the left mouse button down 
- Move the more into the click area of the Switch 
- Release the left mouse Button 

I guess something like that happens when double-clicking too fast, e.g. the click area being disabled for a brief moment (like propagation delay) and the click area only captures the mouse-up event.

richardabendroth, 2021-06-02:
Sorry just read that you already posted that 🙈

closed 2021-06-10 by flandreas