flandreas/antares #139: Feature Suggestion: De Morgan Gate Equivalents

richardabendroth, 2021-05-31:
Related to #113 it would be practical to have a way of converting a gate to its logical equivalent based on De Morgan's Gate Equivalence. E.g. on an AND gate this could be done via right-click on the component in the circuit which offers an option 'Convert to Equivalent NOR Gate' that then has both inputs inverted, replacing the original gate in situ. 

[Nuts & Volts, Aug. 2007, Marston R., Figure 12](https://www.nutsvolts.com/uploads/wygwam/NV_0807_Marston_Figure12.jpg)
Missing on the image: 
- Buffer with inverted Input = Inverter 
- Inverter with inverted Input = Buffer

Kitick, 2021-05-31:
That could be interesting, I actually didn't know thats what it was called, I learned that from my computer science courses called the rule of inversion. Where in a logical statement, inorder to invert the output, you switch everything to it's opposite. So > to <= and vice versa, and in this case AND to NOR, and OR to NAND.

flandreas, 2021-06-01:
This would require #113 to be implemented first, right? Once #113 is implemented, the user can fulfill the requirements of this feature #139 by replacing the gate and negating the inputs by hand. So #139 is an interesting "productivity" feature, but somehow low prio. Let's do the things the user can't do at all first.

richardabendroth, 2021-06-01:
From my perspective as an early-stage learner coming from a non-electronics field, the hardest thing is to deal with boolean algebra at the gate level. For me personally, this would help a lot, not for productivity but for understanding and learning by trying things in different ways circuits can be set up to achieve the same outcome. Sure, the truth tables are somewhat helpful but very basic. I am practically using notepad all the time to write down truth tables and then google my way through to the matching gate type. Circuits can become quite overwhelming with a lot of different gates and inversions in sequence and it would be very helpful to have something in Antares that helps in learning the heuristics and rules of discrete logic by trying and playing assisted by the application. This is the notion where my suggestion from #115 is coming from as well.

Kitick, 2021-06-01:
I totally see your point, and for me I come from a background of advanced computer science, so Boolean logic is ingrained in my head. So I guess that's why I tend to suggest a lot more abstractive suggestions because I'm already at a point where I want the program to do more work because I already understand the floor logic.

So I feel like we just have to have ways for the user to be able to see all the logic if they want, or just rely on the abstraction if they already understand it. That's why I really like the idea of all the integrated circuits are actually just built out of the gates and put into an IC and you can see them and see how they work if you want to.

And again, coming from a computer science background, I understand how to abstract layers really easily, so building more complex circuits like mini computers arnt too complex for me. But I totally see your point, for early learners we should be careful how much abstraction we put into it. But that goes back to what I said in #129, we just need to decide how much abstraction we want, and I'd we do it should be at the users discretion of if they want to understand it or not. My viewing the sub circular or whatever else.

closed 2021-06-24 by flandreas