flandreas/antares #140: Suggestion, implement real ICs?

Kitick, 2021-06-01:
I got this idea from some of the other discussions, when simulating real circuits was mentioned.

So what if there was a library that contains some of the real chips like from the 74 family that are exactly how they work in real life? Their propagation delay is fixed, to what that chip actually performs at, and it only works if it's connected to 5V and ground. Or whatever voltage the chip runs at. It could teach you how to build real circuits with breadboards, and if you over voltage it could kill the chip And if you undervolt it it could just not work.

That's another thing, what if there was a breadboard image that you could snap components like that too, so you could literally build a simulated circuit from real life. The hardest part with that I would think would be implementing the power rails.

I do understand actually we could do the chips without adding any special code, we just make ICs the way you do normally and just name them those chips. Though the propagation delay would be able to be set, and that might add some unrealism to it.

This kind of goes back again, I feel like there should be a analog mode to the program, because a lot of the things that are being talked about are conflictions between how a analog circuit program works and how a digital circuit program works. Do we want to have it be both, have modes to go from one or the other or how does all that play into the whole program?

flandreas, 2021-06-02:
Thank you for this suggestion. My view on this is as follows.

Although I'm fascinated by Ben Eaters "Let's build a breadboard computer from scratch" videos, in which he does a great job in explaining how they work, I'm personally not really into breadboards, because IMHO, they don't help to understand the system, they rather complicate things. Breadboards and 74 family pin layouts are made for the real word, not for explainability.

However, I understand that some users see this differently, and I don't preclude that Antares will some day contain features that support these things. But currently, they have low priority for me.

As you say, there is nothing that prevents you from building your own library of 74 family components, besides only that an Antares project can only be based on a single library, not many libraries. And sharing library files is somehow cumbersome. This is where Antares Web will bring much more functionality.

Regarding "voltage control and overpowering circuits", I think that an even more important aspect is timing and fan out. Timing is already covered well, but fan out control is completely missing. Fan out is perhaps something to be considered implementing in future versions. And propagation delay: If you build your 74 family chips from basic gates (or even from transistors?), the propagation delays are naturally derived from those of the basic gates, so they should be accurate, right?

richardabendroth, 2021-06-02:
I build my 74 circuits for Antares from the basic gates. If you want to add more realistic behavior to them, Antares lets you chose the 'footprint' freely and you could additionally add inputs for VCC and GND and connect them to a set of tri-state buffers or gates with zero propagation delay on all the other inputs so the chip is dormant when VCC or GND are not connected to the VCC or GND components from the net section of the standard library. The only thing Antares cannot simulate right now is different propagation delays depending on voltage and underpower conditions. 
The thing that is really missing is a way to exchange components and libraries and work on them together with others. I'm not too sure that this is something that only relates to the online version as I think this should be an integral part of the offline version as well. A library could be just a public GitHub repo that follows a defined directory and branching structure and can be subscribed to by pasting a link instead of opening a local file. Those links could just be listed on the Antares Website. Antares downloads the content and imports it as a library. On startup, Antares would check if there is a newer version of the 'main' branch for every library and then re-download it and update the files that have changed.

Kitick, 2021-06-03:
That's all fair, and honestly I thought of that as I was writing it, just to make mockup versions with the components and just use AND gates or as you said tri state buffers to simulate VCC and GND. Also that's a good point as well, giving a transistor more voltage makes it switch faster, so the propagation delays would need to be dynamic based on the current voltage. And yes the delays would be totally based on the gates, and simulated normally. But I do agree, this is something that is more of a niche thing, and are more for practicing real development rather than conceptual understanding. And I also watched Ben Eaters build a computer, that's where I first got into circuits actually heh, small world. Thank you both for the input though!

richardabendroth, 2021-06-11:
In the light of #152 possibly being implemented, at least for the Voltage/Propagation Delay relationship you could then have a generic property that is 'Vcc' or sth that is then used with a formula within the propagation delay property of all gates within the component. So you could enter sth like a formula within each Propagation Delay: 
$MAX(90, $MIN(15, 23-(6*ln($('Vcc')-2)))))
where 
- $MAX(a, b) and $MIN(a, b) are function expressions that take two arguments and return the mathematical maximum or minimum of those arguments respectively and 
- $('Vcc') is a reference to the value of the Vcc generic property 
This would adjust the propagation delay according to the voltage entered in the Vcc generic property of the IC Component. 
@Kitick does that make any sense for what you are trying to achieve? 
@flandreas I'm just making up syntax here for illustration purposes. I bet there are nice expression parser libraries out there with much better syntax than what I just made up. 

Also, if Antares would allow to 
- have an 'Active' property for each component or gate, that is default true but can be filled with a formula and 
- have access to the pin status (0, 1, undefined/Z) you could also implement the IC being dead as long Vcc is not 1 and GND is not 0
Example for formula in the 'Active' Property: $AND($('Vcc')==1, $($('GND')==0)

Kitick, 2021-06-11:
Interesting how you use scripting modifiers like $, what languages have you studied? Looks like SQL or Ruby, but thats just a guess.

Anyway, I sort of see what you were doing. What I was going for, was that if a chip (or gate, component, whatever) has a rated voltage, then you would want to give it its rated voltage. If you give below the minimum, then the transistors either switch in a really long time, randomly, or not at all. And as you give it more voltage between the minimum and maximum, then it works faster. As real transistors would. However, going above the maximum could cause them to switch unstably, or kill the transistor altogether.

It's kind of how overclocking works with modern components, you can technically go beyond the rated max, but your playing the silicon lottery that your specific chip can overvolt more, thus giving faster clocks, and more performance. Now thats a little more complex of an idea and probably less of what your looking for, but that was what I was talking about.

@richardabendroth did that answer your question?

richardabendroth, 2021-06-11:
Yes understood. That is far away from what I was suggesting 😅

closed 2021-06-24 by flandreas