flandreas/antares #141: Idea for programming computers

Kitick, 2021-06-02:
Quick thing, sorry for opening so many more discussions lol!

But anyway, so as was mentioned in #136, if you wanted to build and program a computer, like I have a crude one, you get to a point to where you have to program it. With one, its microcode, and two, the instructions. They need to be very precisely handled so that they are interpreted and work correctly. And of course with simple computers you can just use static EPROMS to handle the combinatorics. But my idea, and its kind of out there, is what if Antares had a programmable compiler?

Now hear me out, in my computer, I made a simple little script in JavaScript to convert basic line instructions to my Antares computers HEX codes so that I could write more complex programs faster. Well, even it has limitations because well, modern compilers are very complex when you unpack all the things they do. In any case, what if Antares had a basic, rude assembly style code language that you could set it to convert into a string of HEX codes that can be put into an EPROM? Of course the microcode still needs to be done manually, or maybe that could be helped to? Like there's a mode to calculate a sequence, then by selecting what wires are high and low at a state, and it gives you the HEX code to make that combination.

I went though a lot of stuff, and even if it were to be implemented it would be in the future for sure. just let me know your thoughts on that when you get a second!

richardabendroth, 2021-06-02:
I have been thinking about this as well before but ended up in a loophole of a trillion questions: 
Wouldn't it need to be a completely different type of compiler (or actually an assembler) for every different type of computer someone builds? For example, if someone builds a computer with instruction sets, the internal addressing will always be different, depending on which module is connected to which control wire. Some computers just have a simple ALU with one or two functions, some will have more complex ALU functions. The instructions in the microcode would look completely different. Some computers have a fixed number of cycles for all instructions, some have an instruction terminator. Word length is different for every computer also depending on the bit-width of address and data busses. NISC or Single-ISC computers don't even have microcode in the classical sense. 
What could maybe be a first-ish approach would be some sort of a way to store the program in a property of a ROM and when a simulation that includes that ROM is started, it takes that code and runs it through a user-defined script (the assembler) that is also stored in one of the ROM's properties and writes the output into the ROM.

Kitick, 2021-06-02:
That's what I was thinking that it could do, as I said a sort of, programmable assembler. Because yes, of course any computer built will have custom microcode, that's even true for today's motherboards. And every processor is built to support the specific instruction set(s).

What I'm more saying, is Antares could have a basic layout of a assembly language, like basic instructions, but the user can set in a file what opcodes they get converted to. And as you said as well, some computers are fixed instruction length, and some have terminators. Some are just programed to not even need them (So the start of another is the end of the last). You could set what code each instruction converted to, and the compiler is simply doing the conversion and organization for you. Organization in terms of sub routines, the stack, etc.

Though getting a compiler to build programs with sub routines could be difficult, I'd at least say basic linear conversion would be a good start.

Kitick, 2021-06-02:
I technically made a compiler for my computer, and just hard coded the microcode, and sent a string as a parameter to the function, and it returned a string of hex code separated by a space. And that was the file that I upload to my EPROM. But all it did was convert and space the instructions properly. It also just assumed everything was decimal and converted it to hex. Theoretically an assembler would want to have ways so you could enter binary hexadecimal or decimal numbers with a # $ or x b as is common.

With that example, once you had that kind of program then all you have to do is just have a way for the user to put in their own op codes and conversions, and then there you go you have what I'm proposing.

flandreas, 2021-06-07:
You may not be aware, but Antares already has a "Disassembler" property on the ROM, where you can put a script that disassembles machine code and displays it in the ROM content view. The assembler would be kind of analogous.

However, it feels like kind of cheating to allocate this functionality on the digital layer, and I think I would rather wait until Antares support higher abstraction layers and allocate this feature then on the "Assembly Language Level".

closed 2021-06-24 by flandreas