flandreas/antares #143: Application Hangs due to wiring algo going wild

richardabendroth, 2021-06-02:
Given: 
[image]

Move lower transistor up: 
[image]

Try to move it back down (or somewhere else) and the application will hang for several seconds while doing some rocket science math in the background.

richardabendroth, 2021-06-02:
And refuses to close: 
[image]

richardabendroth, 2021-06-04:
@flandreas can I help somehow to investigate this issue?

flandreas, 2021-06-04:
@richardabendroth That's kind, thank you! I hadn't time to look into this one. I'm still stuck with #150, which turned out to be tougher than I hoped. It requires quite some changes in the start-up procedures of the simulation, and I have to be careful not to introduce regression problems. But alas, I hope that this one will be easier :-)

flandreas, 2021-06-07:
Antares is suddenly starting to draw polylines with quite a lot of points, thereby eating up all your heap memory. Sorry for that :-)

2021-06-07 16:40:20,072 TRACE [AWT-EventQueue-0] (EdgeViewLineStyling.kt:30) - Draw polyline with 2 points
2021-06-07 16:40:20,072 TRACE [AWT-EventQueue-0] (EdgeViewLineStyling.kt:30) - Draw polyline with 262147 points

richardabendroth, 2021-06-07:
That explains it 😅

closed 2021-06-10 by flandreas