flandreas/antares #144: Feature Suggestion: Configurable Switch Output

richardabendroth, 2021-06-02:
Currently Switches always output a logical 0 or 1 depending on their state. For learning about pullup/pulldown resistors it would be nice to be able to set Switches to be high impedance (H) for off. There are two ways that I can imagine: 
- Have a property on the Switch that can be set to 0/1, H/1, H/0 or 
- Have a Switch with an input and an output that will always be H when it's in off state and when in on state will just act like a wire in between input and output.

flandreas, 2021-06-02:
You can do this already now on a circuit input. Press the ALT key while clicking with the mouse, or enter Z when the circuit input has the focus. I introduces this when I develop transistors and pull resistors :-)

richardabendroth, 2021-06-04:
I just realized I made a mess out of the suggestion. Actually, those are two suggestions to the existing standard button: 
- Being able to set the output behavior of buttons: what is output when the switch is dark-green, what is output when it's light green. This enables the beginner user to use a button for active low and active high inputs without gates in between. 
- Being able to set the button as having two input/output ports instead of one output port: it would be great to have a straightforward way to have a button 'break' a wire and output Z to the subsequent net much like a real button does (and like the one I posted in #150 that relies on something that only exists in the simulator but is impossible electronically). A good beginner's usage example (which I took from my recent learning experience) is to learn about pull resistors connected to buttons on circuit inputs to prevent floating inputs. In my honest opinion, this is something every learner will want to understand at some point.

flandreas, 2021-06-07:
I think you are asking for new built-in component "Real switch", something other tools support as well.

richardabendroth, 2021-06-11:
Until then, for anybody who might need it, this can be simulated by creating a component with the following circuits: 

Button (momentary Switch): 
[image]
[image]

Switch: 
[image]
[image]

In Action: 
[image]

flandreas, 2021-06-13:
Great, thank you!

I was also thinking about whether "execution scripting" could be used to achieve the same without a complex inner subcircuit, but that would require triggering re-execution of a subcircuit when a button gets clicked, which is not yet possible. And it would of course only work in "shallow simulation" mode.

Maybe a "pure scripted component" would be useful in the future?

A sidemark: I've noticed that you place the blue crosshairs marker at the corner of the box. I usually put it at the end of a pin, so I can take advantage of the alignment feature when placing parts to be connected.

richardabendroth, 2021-06-14:
I'm not too sure script-only components would be too good as it somehow contradicts the one point of how Antares pushed me to learn. The ability to see the inner workings of things (even into the basic gates with the Mnemonics) and also the requirement to really build a component from gates instead of going the easy way of just scripting them pushed my learning immensely. A good example of that is the Pulse Generator. Really easy to script but the circuit contains a very fundamental concept. Again, I'm very uncertain of this point as those script-only components enable things that are hardly possible to achieve with gates. 
Maybe this can be boiled down to the following: 
- Basic Components (Switch, Pull-Resistors, Transistors, Gates[?], ...) that are hard-coded and could potentially be script-only components that could also contain some sort of visual part (e.g. user-made switch with visual state and state-related icon, onClick Events, Explorer Icon etc.)
- Component Circuits as they are now that are made from Basic Components and other Component Circuits
You could potentially open the Basic Components to userland and allow users to create those components using scripts and visuals. The question is how far this can/should go: i.e. should there be userland inputs/wires/splittes/etc. or are those a third type of Component that is generally non-userland? 

Regarding the blue crosshairs: Nice! Thank you! I've always wondered where to put it 😅

flandreas, 2021-06-15:
I basically agree with you. I just thought that the "controls" feature could be leveraged in a way that the user can use built-in components for the UI (such as switches), and just provide a little bit of script code to come up with a totally new functionality. Your "real switch" serves a nice example for this opportunity: One switch, two bi-directional inouts, and a little script that would be triggered by the switch to produce the output signals. But of course, your full-blown circuits is far more comprehensive in terms of learning.

closed 2021-06-24 by flandreas