flandreas/antares #145: RGB LED uses wrong format

Kitick, 2021-06-03:
I was playing around with the RGB LED, and quickly noticed that the light takes BBGGRR in that order (HEX), when usually colors are represented as RRGGBB. Its not that major of an issue of course, its just confusing because any color code you use has to have the red and blue flipped.

Kitick, 2021-06-03:
Also, when you start the simulation, the LED is red, instead of black. Actually this seams to just be the 000000 state

Kitick, 2021-06-03:
One more suggestion, could there be a way to make the LEDs, both RGB and the normal one be square instead of circle? I wanted to play around and make a little display for my computer, and wanted to use like a 8x8 grid of square RGB LEDs, and use a EPROM and a very fast clock to 'draw' an image provided all the color codes are listed. And it would look better if they were squares.

flandreas, 2021-06-03:
Thank you for reporting this. You seem to be the first to use RGB LEDs :-) I'll fix as you proposed.

A question regarding the square shape: I assume in this case you don't want to have the non-luminous border, right? So you can build an entirely luminous area.

flandreas, 2021-06-03:
[image]

Kitick, 2021-06-03:
Ooh yes I didn't even think of the border! Yes in this case not having the border would be good, just a colord square, essentially a pixel lol. I'm glad I was able to give this feature some attention, and the improvements were apparently easy to implement.

I just thought of this sorry, but what if you can make them smaller? It's not a issue at all, I'm fine with my grid as I have them, but if I wanted to make bigger grids, like a 64x64, the size just to show the LEDs gets large. Actually I don't even know if Antares could handle that many registers connections and LEDs being simulated at once, it was having some proformace issues first starting the simulation with even a 4x4. But after the first start it was fine.

But anyway it seams I might have a cool little display soon! The hardest part is I need a register PER led so that it holds it's current color, and that's fine, it's just a lot of registers. And I don't think I could use RAM, because I need the output constantly held to the LED.

flandreas, 2021-06-03:
Feel free to open another issue just for the smaller LED size.

Regarding performance: When simulation is started, the entire graph is expanded, setting up sub circuits recursively. With large and complex circuits, especially with lots of sub circuits, this takes time. You can always speed up simulation by setting simulation speed to the most-right position, and deselecting "Deep Simulation", which then uses execution scripts instead of fully expanded sub circuits.

closed 2021-06-10 by flandreas