flandreas/antares #147: Input not touchable

richardabendroth, 2021-06-03:
In a complex shape, one of the symbol inputs cannot be connected to anything (does not show the yellow bubble when hovering). 
[image]

flandreas, 2021-06-03:
Interesting. I can reproduce it, but in my case, it is the pin pointing south that can not be connected.

richardabendroth, 2021-06-03:
Maybe it depends on how you draw the shape or which of the pins you place first?

flandreas, 2021-06-03:
Yeah, probably the order of placement. I'm investigating... :-)

flandreas, 2021-06-03:
Found it. The algorithm looks for the pin with the minimum distance from the mouse location, and used the location of the pin as a reference instead of the connection point at the end of the pin line. You are probably the first to place two pins at the same location ;-)

Thank you for reporting this!

richardabendroth, 2021-06-03:
You are welcome! I was learning about Transmission Gates which might be the only symbol out there where this case even appears 🙈

closed 2021-06-10 by flandreas