flandreas/antares #148: Error when starting Simulation

richardabendroth, 2021-06-03:
```
Version: 0.6.0
Start creating new EdgeView at PortView
Connect output of Switch with input of Tri-State Buffer
Move component 'Switch' with ID 1 by mouse
Enter execution mode
Enter edit mode mode
Open application data
Change property 'Toggle' of component 1 to 'false'
Open application data
Add Component 'Switch' from library/project
Add Component 'Power' from library/project
Change property 'Orientation' of component 9 to 'South'
Move component 'Switch' with ID 9 by mouse
Start creating new EdgeView at PortView
Connect output of Power with input of Switch
Add Component 'Pull Resistor' from library/project
Start creating new EdgeView at PortView
Connect output of Switch with input of Pull Resistor
Add Component 'Probe' from library/project
Start creating new EdgeView at PortView
Enter execution mode
java.lang.ClassCastException: class ch.scorpion.antares.model.g.r cannot be cast to class ch.scorpion.jabbah.graph.model.A (ch.scorpion.antares.model.g.r and ch.scorpion.jabbah.graph.model.A are in unnamed module of loader 'app')
	at ch.scorpion.jabbah.graph.model.d.f.resolve(SourceFile:152)
	at ch.scorpion.jabbah.e.j.a(SourceFile:71)
	at ch.scorpion.jabbah.e.j.a(SourceFile:60)
	at ch.scorpion.jabbah.e.r.a(SourceFile:63)
	at ch.scorpion.jabbah.e.m.a(SourceFile:31)
	at ch.scorpion.jabbah.e.a.a(SourceFile:43)
	at ch.scorpion.jabbah.e.a.a(SourceFile:28)
	at ch.scorpion.jabbah.graph.m.a(SourceFile:228)
	at ch.scorpion.jabbah.graph.model.f.r.a(SourceFile:217)
	at ch.scorpion.jabbah.graph.model.f.r.a(SourceFile:203)
	at ch.scorpion.jabbah.graph.model.b.a.a(SourceFile:147)
	at ch.scorpion.jabbah.graph.view.N.a(SourceFile:1093)
	at ch.scorpion.jabbah.graph.view.P.invoke(SourceFile:1045)
	at ch.scorpion.jabbah.base.b.a.b(SourceFile:73)
	at ch.scorpion.jabbah.d.e.o.b(SourceFile:1344)
	at ch.scorpion.jabbah.graph.a.k.invoke(SourceFile:1089)
	at ch.scorpion.jabbah.base.Q.run(SourceFile:22)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

```

flandreas, 2021-06-03:
Same as #119 and #130. Already fixed for 0.6.1. Time to release the release... :-)

richardabendroth, 2021-06-03:
Okay, I'll stop posting Errors that have the same stack trace like this one :)

flandreas, 2021-06-03:
Easy, no problem :-)

closed 2021-06-10 by flandreas