flandreas/antares #149: Rookie Question: Why does a pull-down resistor go Z when connected to Power?

richardabendroth, 2021-06-03:
[image]

flandreas, 2021-06-03:
The way how  pull resistors work (at least in my understanding) is that they pull a wire to the desired level if the wire carries an undefined signal. In all other cases, they remain passive, i.e. don't participate in asserting signals to the wire, i.e. are undefined.

But I am aware that this can look confusing. I've got an issue on my hidden agenda for thinking about whether the pull resistor should draw itself differently in this situation, e.g. using a color gradient from 1 (bright green) to 0 (dark green) in your scenario. What wouldn't work is to draw it in 0 (dark green) only, because you would then have an unexplainable clash of 0 and 1 at the pin where the resistor is connected to the wire.

richardabendroth, 2021-06-03:
Ah, I see! Thank you for your explanation! That makes total sense!

closed 2021-06-10 by flandreas